prusa3d/Prusa-Firmware #2384: 3.8.1 Z-height adjust long push during "Heating" fails on MK2.5 or MK3

Astrobirder, 2019-12-24:
I just updated both my MK3 and MK2.5 (both stock) to 3.8.1.  I print from 2 RPi's running Octoprint.  Previously, when I started a print, I could do the long button push, and push the Z-height up to make it easier to remove ooze before the purge and mesh bed leveling.  

Now, the long push is ignored, and the menu doesn't have the Settings option as visible--just Info, Tune, Fail stats & Support.

ochm, 2020-01-19:
IMHO this behavior is a feature, not a bug.

Panayiotis-git, 2020-01-19:
During the printing, the user is not allowed to move the z axis. But unfortunately, the pre-heating at the start of a print is also considered as part of the printing.

closed 2023-10-12 by github-actions[bot]