prusa3d/Prusa-Firmware #2392: No M117 text dispalyed after Self test

ochm, 2019-12-30:
**Printer type** - MK3S
**Printer firmware version**-  3.8.1

**Describe the bug**
When printer finished Self test (started from menu) the text **Self test OK** is displayed,
but from now, display does not show text sent by M177, still is displayed  **Self test OK**
  
**To Reproduce**
Do Self test, when finished, try to send M117 whatever 
from some console, for example Octoprint terminal.
  
**Expected behavior**
After selftes display what is sent by M117

3d-gussner, 2020-01-02:
I can confirm the misbehavior. @ochm Thanks for pointing out.

gudnimg, 2023-07-12:
Tested this on my MK3S+ and can confirm the issue is gone. Probably fixed in 3.11 or 3.12.

closed 2023-07-12 by gudnimg