prusa3d/Prusa-Firmware #2396: [BUG] Extruder 1 is enabled after any error detected

anilrobo90, 2020-01-03:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
MK3S
**Printer firmware version**-  [e.g. 3.8.1, 3.8.1-RC1, ...]
3.8.1

**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]
MMU2S
**MMU upgrade firmware version [e.g. 1.0.6, 1.0.6-RC2, ...]
1.0.6

**Describe the bug**
After an MMU crash or Power panic recovery, if manual intervention is required to resolve the MMU issue, following a reset, the extruder 1 is enabled by default even if before the crash extruder 5 was being used.
  
**To Reproduce**
Slice any object with full soluble supports profile in Prusa Slicer
during the print remove the filament from Extruder 5 this will cause the printer to crash, 
After the crash is detected please feed the filament back and reset the MMU board.
You will now notice the extruder is on position 1 instead of 5.
  
**Expected behavior**
After a crash, the same extruder should be enabled as it was before the crash.

vintagepc, 2020-01-03:
I think this gets into a broader persistence of current extruder issue. Anything that ends up resetting the MMU loses the selected state and can lead to some nasty situations. For example, I've had my system get stuck trying to retract/unload filament 1, but it was actually filament 3 that was loaded, and the MMU never pulled it out. 

I don't have a sense of how practical or risky it would be long term in terms of states getting out of sync, but it would be very nice to have the MMU recall the selected filament (or be told it by the Mk3) across power cycles and similar events. I like to leave a given filament loaded for many days and do a batch of prints sliced without single mode or MMU codes. I would use single mode far more often if it had the capability to just leave the single filament in the nozzle and not unload/reload/do an extensive prime every time if the new print uses the same one, especially owing to #2069

vintagepc, 2020-01-09:
Just realized this is closely related, if not a duplicate of #2159

lew-robobionics, 2020-01-15:
@vintagepc #2159 relates to power faults.

This one relates to all the MMU faults

vintagepc, 2020-01-15:
> @vintagepc #2159 relates to power faults.
> 
> This one relates to all the MMU faults

Yes, I'm noting it here for cross reference/housekeeping. The fix for one vs the other is going to likely be very similar or identical, the others can also be tested/closed.

closed 2023-10-16 by github-actions[bot]