prusa3d/Prusa-Firmware #2398: Feature request: Add option to beep when finished print

mrjohnc, 2020-01-05:
Add option in settings menu for the printer to beep when a print has finished (like a microwave does). This will help to get the attention of the user if they are not in the room, listening to music etc.

mrjohnc, 2020-01-05:
Thanks very much :)

mrjohnc, 2020-01-05:
> > > Hello John,
> > > Thanks for your feedback!
> > > Actually, we have this for long time in our list and still thinking how to do it to satisfy all people.
> > > We'll definitelly do that!
> > > Michal Průša, CTO
> > > --- original message ---
> > > > Add option in settings menu for the printer to beep when a print has finished. This will help to get the attention of the user if they are not in the room, listening to music etc.
> > > > --- end of original message ---
> > 
> > 
> > Hello Michal,
> > I'm looking forward to see this option implemented.
> > Do you have any news about it?
> > Thank you in advance,
> > Andrea
> 
> Dear all,
> there is already a super easy way to do it.
> Edit the Slicer setting and let Slicer add some G-Code at the end of all generated G-Code files which does the beeping. I would recommend the following snipped:
> 
> ;this snipped plays a kind of imperial march
> ;great for the end of your print job :-)
> ;imperial martch
> M300 S220 P300 ;LA3
> G4 P300
> M300 S220 P300 ;LA3
> G4 P300
> M300 S220 P300 ; LA3
> G4 P300
> M300 S174.61 P350 ;F3
> G4 P75
> M300 S261.63 P250 ;c4
> G4 P100
> M300 S220 P350
> G4 P100
> M300 S174.61 P350 ;F3
> G4 P75
> M300 S261.63 P250 ;c4
> G4 P100
> M300 S220 P400
> 
> ;-)

I didn't realise it can play songs 👍

3d-gussner, 2020-01-08:
@mrjohnc If you got an answer to your issue and it is working for you, can you please also close issues that have been solved?

mrjohnc, 2020-01-09:
> @mrjohnc If you got an answer to your issue and it is working for you, can you please also close issues that have been solved?

To the best of my understanding this feature request has not yet been implemented yet

3d-gussner, 2020-07-13:
@mrjohnc There is no need for a firmware change as you can add beep, music in your end gcode.
Please close this issue.
[stale][fixed by end-code]

mrjohnc, 2020-07-13:
This depends on if the person wanted it to beep a certain number of times or if they wanted something like 'beep until button in pressed.

My understanding is that this issue should be moved to the PrusaSlicer to implement as a software feature, @stelgenhof prusa3d-bb is there a way to move this issue across to the other repo or does it need to be recreated?

It could be an option where you tick an option called something like 'beep at end of print' and offer an option to paste in 'music' or give a little tool tip for how to do this with an existing field.

stelgenhof, 2020-07-13:
@mrjohnc I don't think it is possible to transfer issue automatically between repositories. At least that I am not aware of :)

rdragonrydr, 2020-07-13:
I would personally have preferred this to be a setting I could enable or
disable globally on my printer rather than having to re-slice everything,
but having a default option and a checkbox in the Slicer *would* be an
acceptable solution.

mrjohnc, 2020-07-14:
So this should stay as a hardware request? 

I guess this is difficult to know where to put because there are two ways of implementing it...

rdragonrydr, 2020-07-14:
Hardware was my preference. I was hoping for a "none, beep, or _keep_ beeping" selector in the onboard menu that I could set up and leave depending on where I had the printer. (And of course this would apply to all prints rather than just whatever new ones I make going forward. Some of my stuff needed fiddly slicing and I don't want to have to redo it all, since some of them are for projects that I keep making more of.)

Of particular note is the option to keep beeping until I go collect the print, because if I were to just add it in the end G-code it might end before I notice or keep beeping even after I get to the printer (since it would literally be a "beep x times" command rather than allow user interaction). With a hardware request, it could listen to the button status and shut off when I press the button to acknowledge.

bszonye, 2020-08-08:
End code is not currently an option for Prusa MINI owners, as the firmware does not appear to support M300 yet.

bszonye, 2023-08-14:
Looks like M300 still isn't fully supported on Prusa MINI (see prusa3d/Prusa-Firmware-Buddy/issues/334), plus that's still only a workaround for the requested firmware option. Still would like to see this.

3d-gussner, 2023-08-30:
@bszonye This is the repository for MK3/S/+, MK2.5S. Please open an issue for the MINI here https://github.com/prusa3d/Prusa-Firmware-Buddy
Closing issue.

closed 2023-08-30 by 3d-gussner

bszonye, 2023-08-30:
No need, the issue that I linked above already covers this for MINI 👍 got mixed up about which issue was in which repo, sorry

rdragonrydr, 2023-08-30:
Can I _un_close this? The feature request was for firmware support for a beep on completed print.

The "workaround" solution of using end gcode has the issue of being a finite length _and_ of locking up the printer until it stops beeping. It also does not work (persistently) if using the factory presets, as those cannot be saved or edited.

I would have to make a preset for each temperature, infill, etc _and still edit the end gcode_ for each, then the printer would beep a few times and _stop_, so if I was not paying attention I would never know the printer was done. Alternatively, if I'm monitoring it, I can't start the next print until the beep cycle completes, so I can't make it indefinitely long.

TL;DR: Gcode cannot meet the need for a feature that can beep indefinitely or that does not lock up the hardware UI while beeping. I want something like the filament change alert **that already exists**, but that turns on when the print is done (just... without ejecting the filament).