prusa3d/Prusa-Firmware #2400: [FEATURE REQUEST]<Guide to Alter Settings for Optimal Filament Tips for MMU2S>

lew-robobionics, 2020-01-06:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - MK3S MMU2S
**MMU Upgrade** - MMU2S

**Is your feature request related to a problem? Please describe.**
 I'm always frustrated when I have to manually run through a filament for almost 20 times before I can get a near optimal tip so that the MMU Doesn't crash.
  
**Describe the solution you'd like**
  A standard Gcode like first layer calibration in the Menu to tune settings of loading, unloading and ramming for using the MMU2S in the best way possible.

vintagepc, 2020-01-06:
I think you'll have more luck posting this to the PrusaSlicer issue pages; the printer firmware is too low-level and does not deal in those parameters, it only executes the resulting gcode moves that PS generates from them. But interesting idea, would be nice if PS could have a function to generate a series of changes within specified parameters, from which you could then select the best-shaped tip.

3d-gussner, 2020-01-08:
@lew-robobionics & @vintagepc Good idea, maybe something like http://marlinfw.org/tools/lin_advance/k-factor.html for the MMU tips would be helpful.

jokohoko, 2020-01-18:
What settings do you change in between the attempts? I have the same issue of bad tips, but am stuck on what can be changed. So far i only got small changes on the tips and never an ideal setting.

lew-robobionics, 2020-01-20:
@jokohoko - possibly the number of cooling moves and speed of load or unload

JohnnyDeer, 2021-02-03:
Hi @lew-robobionics, @vintagepc was right this issue is for PrusaSlicer, please opent it there.

closed 2023-10-16 by github-actions[bot]