prusa3d/Prusa-Firmware #2401: Z axis calibration/movement issues MK3s - Firmware 3.8.1

Tanischacat2009, 2020-01-06:
I am not sure if this is a firmware related issue or not, but a lady from Prusa Technical support suggested that I revert to an earlier firmware version and post here.

I just bought and assembled a new MK3S printer within the last couple of weeks.  Upon putting it together, I had ran into a number of issues of getting it to calibrate.  This is the second one of these that I built and didn't have the same issues as this one and much fewer too for that matter.  Anyways, I had flashed the latest firmware versus the 3.7.2 version that the printer had shipped with before even trying to Calibrate to possibly avoid having to do it twice.

After many hours of trying and attempts, I finally got the printer to pass all calibrations successfully, including ZYX, however, when I go to use the printer, it chooses when to work and when not to.  When doing mesh bed leveling at the beginning of a print, I will get an error message saying that Z leveling needs to be enforced.  When I go to do that, the gantry only moves up part of the way and stops in various places.  After repeating this process, resetting the printer, calibrating "Z" several times and moving the gantry up and down using software or by hand, I am again able to print.  Sometimes, the printer will allow me to restart prints quite a few times before this happens again, other times not.

I had discussed this at length with two different support people from Prusa yesterday whom the first one thought that it might be the PINDA probe being faulty, but that doesn't appear to be the case and that too has been calibrated and checked several times and appears correct.  For the first person I chatted with, I was unable to replicate all of the issues via video because the printer decided to co-operate and do what it's supposed to do.  Later on in the day, this wasn't the case and I recorded another video for the Prusa support person in which I will share here if needed.

The second person just suggested that it might be a firmware issues and she was the one who told me to post here and try 3.8.0, which I haven't done yet.  She also said to just deal with the issue if I don't find it too annoying, but it is when you have to spend up to an hour trying to start a print and more so if you don't have enough time or get interrupted and have to try again and again later etc.

I am also using 3.8.1 on my older MK3s printer which I built last March and do not have the same issues, so I can't even be sure it's firmware related at all.  My guess if it isn't, it really is the PINDA or another hardware issues like bearings slipping etc.

leptun, 2020-01-06:
@Tanischacat2009 Thanks for reporting the issue. This does indeed sound like a hardware issue, but I can’t say with certainty what it is. There are multiple possible causes:

- A malfunctioning PINDA. It is however quite unlikely since you said it stops on its way up to the top when “Z leveling enforced” happens, which leads me to the second possible cause

- Z nuts on leadscrews are too tight or issues with Z motors in general. When the printer does the mesh bed leveling or calibrations it detects 2 things: the PINDA and the Z motor stall detection. If either of these happens in an unexpected position (or randomly), then that is a clear problem and the printer errors out and aborts the action as to not damage the bed/nozzle.
When you said that the Z axis was NOT lifting to the top, that sounds a lot as if the stepper drivers detect a motor stall before reaching the actual target (leading to the unexpected situation logic) and also that it might detect that the nozzle crashed into the bed when doing the mesh bed leveling (which might or might not be happening) and tries to recover from it, but fails to do so.
TL,DR: Check if the individual leadscrews move really hard or in an inconsistent way.
It is very unlikely that this is a software issue as there was no change involving this part of the fw between versions 3.8.0 and 3.8.1.

If you have a video, that might be able to help us determine what kind of input the firmware got when it triggered the error (and how the printer reacted to it). Were you able to identify any pattern in the failures? Eg: always fails on the right side or something like that.

Tanischacat2009, 2020-01-06:
Thank you, Leptun for your quick response.

I would guess that the issue too is most likely unrelated to firmware and the lady who told me that might have even sent her response to the wrong person as she did once before during the conversation and corrected herself.

My guess is that the issues are related to hardware or assembly issues.  The issue happens at random but yes, it's most frequently on the right hand side, but it's also happened elsewhere.  If it's an issue with the lead screws being too tight, are you referring to the black ones on the bottom just above the motor?  Everything seems to turn easily by hand or when using the LCD menu to move the axis.  When the errors occur, the nozzle does NOT crash into the bed.  The only time this had happened was twice when I mistakenly shut the power button off and on without first canceling the print that was in progress.

Another thing that came to mind is that I had the printer set in STEALTH mode when I calibrated it and the update description for 3.8.1 firmware states that Z doesn't calibrate in Stealth mode, yet it does, somehow, but maybe not properly and obviously doesn't stay that way.

I would think that if one or both of the z motors were faulty, the z axis wouldn't work at all without moving it manually.

Here is a link to the video which MAY work.  Please be patient as it's a large download

https://1drv.ms/v/s!AiBwZeaUaAyagsE_ocbLMJsN4b3sng?e=yaaFqo

Tanischacat2009, 2020-01-07:
I reset the printer to NORMAL mode, re-calibrated Z after resetting the printer.  That worked like some times before.  I continued to try and print but was having other issues with filament adhesion etc. and was trying to work through those.  I also noticed that the print shifted drastically on the second layer (Y axis, never had this before on this printer).  I then decided to re-run all of the calibrations and now I am getting "Y-axis length errors".  My belt does seem to be a bit loose but no worse than before and the number is 240 which is supposed to be good as I understand.  I investigated and couldn't find anything else that would cause the error.  Weird.  I tried tensioning the belt and re-running the calibration again but nothing changed, still the same.

leptun, 2020-01-07:
Leadscrews:
This is the leadscrew nut:
[image]
This is the plastic piece you mentioned:
[image]
Both of these should not prevent the motor from spinning consistently.

Stealth mode: The stealth mode issue should be fixed in 3.8.1. The printer had to switch automatically to normal mode for calibrations but didn't. Now it does.

Z motor issues: Even if one of the motors is faulty, both of them will (mostly) react in the same way.

Try running selftest. The belt tension values only change after selftest unfortunately. This will be addressed in the upcoming 3.9 firmware. (also in the attached zip)

I might have something for you to test that might help with the mesh bed leveling issue. The Z issue still needs to be looked into as not raising the Z to the top is an indication of a stall detected too early, signaling excessive friction somewhere. As for the Y axis, I have no idea what could have happened to that. Let's fix the Z/MBL first and we'll look into that later on.
So for the firmware: I'm working on a fix where for some people using the MK2.5(S) upgrade they encounter the issue very frequently and it is thought to be the fault of interference from the bed heating to the PINDA. As a warning, this is an unofficial firmware compiled by me based on the latest 3.9.0 alpha build with my fix included. You should not leave the printer with this firmware unattended!

[FW381-Build2869-1_75mm_MK3S-EINSy10a-E3Dv6full-EN_ONLY.zip](https://github.com/prusa3d/Prusa-Firmware/files/4029094/FW381-Build2869-1_75mm_MK3S-EINSy10a-E3Dv6full-EN_ONLY.zip)

Tanischacat2009, 2020-01-07:
Thanks again Leptun

I will try your suggestions. 

I did previously adjust the top screws on both sides of the x-axis gantry to "lead screws" as you refer to them as and doing so appeared to make no difference at the time.

As for the Y axis error, I will investigate further as I have a few options to try before refreshing or trying different firmware. The printer currently does not pass the self test hence the Y axis length error. The only hardware issue that I can think of that may have caused this is somehow the belt may have gotten stretched, so, at worst, I may need to take it off and try and tighten it up somewhat.  I won't know my true belt value until the printer passes the self test.

My a axis issues may or may not be resolved but time will tell.

Tanischacat2009, 2020-01-08:
I just went to the trouble to tighten my y axis belt as it was quite loose compared to my other printer.  My belt status is now 266 for the y axis and 246 for the x axis   I then made several attempts to fully calibrate the printer but have had no success. If anything, my z axis issues have resurfaced and seem to be worse. I also loosened all of the lead screws more and nothing has changed. The printer now passes the self test and there are no more y axis length errors or anything else. The printer says everything is ok but fails when attempting dye calibration or just a axis calibration. The gantry continues to stall somewhere in the middle instead of going up and down on it's own. If I play with it long enough, it may do so again, but then likely fail again at some point too  I have recorded a new video of the calibration process.  I have not tried your firmware compilation but will do so next to see if that helps any. More likely, it's the motors like you said or even the board may be faulty.

Tanischacat2009, 2020-01-08:
Video

closed 2023-10-16 by github-actions[bot]