prusa3d/Prusa-Firmware #2406: MMU2 update firmware

lijinlin19970813, 2020-01-09:
I have a question, how do I update the firmware of the MMU2, is there any software to download, please tell me.

leptun, 2020-01-09:
To update the firmware of the mmu you use the same method as updating the firmware of the printer, except you connect a usb cable to the mmu and use the correct hex file (for the mmu).
The manual is your friend. Read chapter 4. http://prusa3d.com/downloads/manual/prusa3d_manual_mmu2s_eng_1_00.pdf?2

lijinlin19970813, 2020-01-09:
> 要更新mmu的固件，可以使用与更新打印机固件相同的方法，除了将usb电缆连接到mmu并使用正确的十六进制文件（用于mmu）。
> 该手册是您的朋友。阅读第4章[。http://prusa3d.com/downloads/manual/prusa3d_manual_mmu2s_eng_1_00.pdf?2](http://prusa3d.com/downloads/manual/prusa3d_manual_mmu2s_eng_1_00.pdf?2)

The first green light of MMU2 is on at 24V, which means that MMU2 is ok. If I use other marlin motherboards to communicate with MMU2, do I only need to determine the serial port? I use the load Filament in the screen ALL or other options, MMU2 has no response.Does this indicate that MMU2 is abnormal?

leptun, 2020-01-09:
No, it indicates that the boards are not communicating. Unfortunately we can only provide support for the Prusa-Firmware, not stock Marlin. As for boards, only miniRambo/einsyRambo are supported. Sorry.

I think that the Marlin firmware had some MMU2 integration option. You could try asking this question on their github.

3d-gussner, 2020-01-09:
@lijinlin19970813 Have a look here https://github.com/MarlinFirmware/Marlin/issues?utf8=%E2%9C%93&q=is%3Aissue+mmu2

There are few reported issues and i am sure also solutions and how-to's

3d-gussner, 2020-01-09:
@lijinlin19970813 Just saw that you already posted something there

lijinlin19970813, 2020-01-09:
> @ lijinlin19970813刚刚看到您已经在此处发布了一些内容

Yes,I also posted the issue in Marlin,but it was not resolved.

3d-gussner, 2020-01-09:
@lijinlin19970813 Sorry to hear that. But as @leptun said this is not the correct place to ask for help with MMU2s connected to something else then a printer with Prusa firmware.
Good luck with that, hope you get it working.

closed 2020-01-09 by michalxfanta