prusa3d/Prusa-Firmware #2407: [BUG]<MMU2s Crash due to false triggering of the FINDA>

lew-robobionics, 2020-01-09:
Printer type - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
MK3S
Printer firmware version- [e.g. 3.8.1, 3.8.1-RC1, ...]
3.8.1

MMU Upgrade - [e.g. MMU2S, MMU2, MMU1]
MMU2S
**MMU upgrade firmware version [e.g. 1.0.6, 1.0.6-RC2, ...]
1.0.6

Describe the bug
MMU Crash due to false triggering of the FINDA.
The filament is fully unloaded but the printer is unable to detect it as the FINDA is not glowing (i.e. false trigger).
It requests for user intervention.
It happens when printing using an enclosure as well as without an enclosure.

To Reproduce
Not Sure

Expected behavior
The FINDA should be properly able to detect a load and unload

vintagepc, 2020-01-09:
This sounds like a hardware issue. If the FINDA is not glowing, your steel ball has not dropped back down and you need to check that. It can't be fixed in software.

lrbrundrett, 2020-01-12:
I duplicated this problem by accident. I was trying to make the Finda less sensitive than the pinda and pulled it out too far. The printer went into endless loops until I moved the FINDA in to make it more sensitive. Now it is working correctly.

lew-robobionics, 2020-01-15:
@vintagepc I've checked various times if there is any stuck debris or filament strand stuck but there isn't any.

So I am not sure why the ball gets stuck.

However if there is a knock on the selector or if the selector is calibrating itself the ball drops, so probably if the MMU detects that an unload is unsuccessful a calibration move before a user intervention would kinda automate a step of troubleshooting.

vintagepc, 2020-01-15:
The problem is the MMU doesn't know that and has no way to identify this specific case accurately. As far as the MMU is concerned, there is still filament in the selector and so it will not move it (for good reason). You're asking to detect a specific case based on the state of a single sensor (which could have many other reasons for being in the same state). I'm going to reiterate that this issue is far better solved by hardware adjustments/changes. 

I'll add a point that even if the FINDA is no longer triggered through automatic action, the system will likely require user service the next time it loads this filament, as it will have unloaded it out of the MMU gears. I'd think it's far better to need attention at the point of occurrence rather than some time in the future; otherwise you'd be left with a system that appears to be randomly unloading too far and unless you sit and watch it explicitly it's not clear why.

I had similar issues with the selector design at one point (does not take much, just a few wisps of scrunched-up filament hair in the right place). I've since installed a user-mod with an open side to help clear debris and a magnet to pull the ball back down; I have not had any further issues with the ball getting stuck. 

Also, could you edit your original issue/title? The term "crash" has very specific connotations in software and you do not appear to be encountering one; the system is operating as it is programmed to, and you're actually asking for a feature/improvement rather than reporting a bug.

lew-robobionics, 2020-01-15:
> I had similar issues with the selector design at one point (does not take much, just a few wisps of scrunched-up filament hair in the right place). I've since installed a user-mod with an open side [...]

@vintagepc Could you please share the link to this? I'll try it out and report

vintagepc, 2020-01-15:
sure, https://www.thingiverse.com/thing:3528314

doug62, 2020-02-29:
There is a software problem with the FINDA, while trouble shooting i pulled it out, showed sensor status and it failed to trigger reliably when metal was placed near it - seems there is a software routine that incorrectly "debounces" signals from the sensor.

lew-robobionics, 2020-02-29:
I reprinted the Idler and the Selector with ABS and the issue seems sorted.

I am guessing when printing PC or ABS using the Lack enclosure the temperature goes high enough to cause expansion of the PETG parts causing too many failures

vintagepc, 2020-02-29:
> There is a software problem with the FINDA, while trouble shooting i pulled it out, showed sensor status and it failed to trigger reliably when metal was placed near it - seems there is a software routine that incorrectly "debounces" signals from the sensor.

The display is notoriously slow to update.

closed 2023-10-16 by github-actions[bot]