prusa3d/Prusa-Firmware #2410: [BUG] [3.9.x] Firmware crash/underflow with small files (when EOF is reached during a move?)

vintagepc, 2020-01-11:
**Printer type** - MK3S (Custom mods: Skelestruder, 0.9 degree steppers on X/Y)
**Printer firmware version**-  Pre-alpha 3.9.x (Github pull Jan 11, 2020)

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**Describe the bug**
The printer firmware crashes due to memory corruption when executing certain small gcode files. I suspect planner corruption when there are moves near the very end of the file, or when a move is in progress and EOF is reached. (Reset does not appear to occur if the final line is synchronous, e.g. a G4 S1)

The attached example causes this reliably on my printer. Playing around with the file causes interesting behaviours. For example, a file containing only the line "G1 X100 F12000" does not move X at all. 
  
Serial captures at the end of the submission. A few times I have even observed a bootloop with "WARNING: STATIC MEMORY OVERWRITTEN" and a loud beep. Pressing the reset key exits the bootloop. (The bootloop happened with an older build, at this point I updated to the latest git pull to see if it had already been fixed)

**To Reproduce**
Execute one of the provided examples.
  
**Expected behavior**
Printer should execute gcode as expected.
  
**G-code**
[MVE.gcode.tar.gz](https://github.com/prusa3d/Prusa-Firmware/files/4049913/MVE.gcode.tar.gz)

The following, even shorter sequence of commands also causes it:
G28 X ;home X
G1 X100 F12000 ;move to center of X axis

**Video**
 Can make one if desired, not much to see though, just the printer homing X, then moving a few mm before resetting.

** Serial log**
echo:enqueing "M23 mve2~1.gco"
echo:enqueing "M24"
echo:Now fresh file: mve2~1.gco
File opened: mve2~1.gco Size: 138
File selected
echo:Unknown command: " "(2)
tmc2130_home_enter(axes_mask=0x01)
  0 step=12 mscnt= 195
tmc2130_goto_step 0 59 2 1000 8
tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
Done printing file
echo:0 hours 0 minutes
Done printing file
echo:0 hours 0 minutes
Done printing file

(Repeats about 20 more times)

Donecho: 3.8.1-2869
echo: Last Updated: Jan 11 2020 16:20:39 | Author: (none, default config)
Compiled: Jan 11 2020
echo: Free Memory: 1726  PlannerBufferBytes: 1792
echo:Stored settings retrieved
adc_init
CrashDetect DISABLED
FSensor ENABLED (sensor board revision: 03 or older)
Sending 0xFF
echo:SD card ok
MMU => 'start'
MMU <= 'S1'
MMU => '106ok'
MMU <= 'S2'
MMU => '372ok'
MMU version valid
MMU - ENABLED

leptun, 2020-01-12:
@vintagepc I was unable to reproduce the issue with both of the provided gcodes. The only issue I found was having no movement if the last line is not terminated with \n (aka the printer waits indefinitely for the line to wait and just sits there doing nothing). Latest MK3 branch (3.9.0-Beta1 before release)

leptun, 2020-01-12:
@vintagepc Can you please provide the hex file that you are using? Maybe it's something related to the 0.9deg stepping (although it shouldn't).

vintagepc, 2020-01-12:
Absolutely, I pushed up the source as well so you can see the diff with Mk3. There are no 0.9 degree changes to the code (all done via M350/M92), only some adjustments to load/unload extrusion distances to account for the shorter distance from the gears to the nozzle. 

https://github.com/prusa3d/Prusa-Firmware/compare/MK3...vintagepc:Skelestruder-MMU2S?expand=1

[FW381-Build2869-1_75mm_MK3S-EINSy10a-E3Dv6full-EN_ONLY.hex.gz](https://github.com/prusa3d/Prusa-Firmware/files/4050632/FW381-Build2869-1_75mm_MK3S-EINSy10a-E3Dv6full-EN_ONLY.hex.gz)

vintagepc, 2020-01-17:
Can confirm I still observe this in 3.9.0-Beta. 

The attached gcode produces a 'STATIC MEMORY OVERWRITTEN' error.

[MVE.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/4079604/MVE.gcode.zip)
[image]

BMac1203, 2020-02-04:
I seem to be having a similar issue. I am running the Bondtech version of the 3.8.1 FW and I continue to get failed prints due to this error :(

vintagepc, 2020-09-23:
Closing, I think this may have been fixed by #2791

closed 2020-09-23 by vintagepc