prusa3d/Prusa-Firmware #2412: [FEATURE REQUEST] Use/Repeat last print (All Prusa Printers)

Tanischacat2009, 2020-01-12:
When using the LCD menu on any Prusa printer and one wants to repeat the last print they used for any reason (making more copies of something or after a failure etc.), a nice feature would be for the menu to have an option such as "use last print" and list the file name of the gcode below it which was used last on the printer, before displaying below the full list of gcode files available in a specific folder on the SD card.

For example, one would select "print from SD" from the main menu on the LCD screen, then one would see at the top of the screen "Use last print?" with the file name just below it, such as "sample.gcode".  Then by scrolling down, one would see the whole list of gcode files as are now.

This is very useful if the file you may have just used happens to be the 50th one in the list.  Another way to do this is to have an option to sort the list according to when the files were used last, versus alphabetically or the time the files were created.

rdragonrydr, 2020-03-30:
I second this idea (I was actually about to add this and the "beep when done" item if not already here).

However, I, personally, think it might be better to have the reprint option as the second item on the main menu (the one that appears when the knob is pressed) rather than require digging down to the print from SD option. This also means that the maintainer would not then need to add a special case to the file-read-and-sort code to have that menu list the last-printed item specially.

In this way, it wouldn't be that the last item to have been printed just floats to the top of the print from SD menu, but instead that if the printer has printed an object **during the same power cycle** (for two reasons: Maintaining the list over power cycles uses more EEPROM, and secondly, if it has been turned off the user is probably done printing that thing), the main menu now has a "Reprint: D&D_Figurine_No._7.gcode" option above the usual settings, calibration, etc items.

leptun, 2020-03-30:
Quick note: should this be implemented, the reprint button won’t contain the actual filename.

rdragonrydr, 2020-04-03:
Fair point. It would probably be too long. I care more about the function anyway, since the user likely knows what the last print was!

sendingstrings, 2020-04-18:
> Quick note: should this be implemented, the reprint button won’t contain the actual filename.

Can you explain why?

leptun, 2020-04-18:
@sendingstrings because there would be very little space for the actual filename. As for scrolling text, I can tell you now that’s not gonna happen. Not enough resources.

sendingstrings, 2020-04-18:
Would it be possible to use another line of the LCD?

leptun, 2020-04-18:
Maybe, but scrolling is still out of the question. I'm not aware of any menu implementation for items that span multiple lines.
What we could do is having a "Reprint last item" button (or something like that) which open a submenu with full filename on the screen and two options: Print and cancel. I think this is the only implementation that is doable.

sendingstrings, 2020-04-18:
> Maybe, but scrolling is still out of the question. I'm not aware of any menu implementation for items that span multiple lines.
> What we could do is having a "Reprint last item" button (or something [...]

That sounds like a great solution.

rdragonrydr, 2020-04-25:
Submenu is fine. Can the default selected option be "print" in that case, though?

rdragonrydr, 2023-10-11:
Any news on this? Does the above solution also support _not_ making another print (yes/no option) instead of resetting the printer or something?

If it _does_ work, can I request it be added as a slicer option so we can check a box to make a re-printable component instead of a one-off?

3d-gussner, 2023-11-29:
Closing this issue as it will be part of FW 3.14.0 and got merged see https://github.com/prusa3d/Prusa-Firmware/pull/4510 

Thanks again to @jfestrada who implemented this.

closed 2023-11-29 by 3d-gussner