prusa3d/Prusa-Firmware #2413: [BUG]<Automatic restart with MMU2 printer>

lijinlin19970813, 2020-01-14:
I use the screen's load filament, load filament 1, the process is normal, return to the main screen of the screen to display “F?” . When I started to print a monochrome GCODE file, the printer was automatically leveled and returned to the origin, and then the printer was automatically restarted. I do n’t know why, and if anyone can tell me what is the reason.

vintagepc, 2020-01-14:
Octoprint? See #2115 . 

No Octoprint? There's a known bug that causes this if the filament sensor reads 1 and the FINDA reads 0. Check/calibrate both your filament and FINDA as per the user manuals.

Note this is not a support area for troubleshooting your printer. For that you either want to use Prusa's live chat or the prusaprinters forum. This area is specifically for reporting bugs and problems with the firmware.

lijinlin19970813, 2020-01-15:
I used the two-color GCODE file downloaded from the official website to print on the screen
[image]
[image]
Why is this wrong?

vintagepc, 2020-01-15:
Where did you buy your printer? Some of the parts do not look like official Prusa parts. If you have a clone we cannot really help you further here; you'll need to get support from the maker of the clone.

Otherwise, you'll need to be more clear about what bug you are reporting or trying to solve. Have you verified your filament sensor and IR sensor are reporting the correct values to address the reset on single-color print?

Your printer appears to be indicating it's trying to change to filament one. We cannot guess what is wrong without a description of what the printer is doing vs what you think it should be doing, so please elaborate.

lijinlin19970813, 2020-01-15:
> Where did you buy your printer? Some of the parts do not look like official Prusa parts. If you have a clone we cannot really help you further here; you'll need to get support from the maker of the [...]

When I turn on the machine, I used the filament on the screen to load 1, 2 filaments. This process is correct, and then return to the main interface will display "F?". I inserted the SD card to print the two-color GCODE files downloaded from the official website. When printing, the heating was normal and the leveling operation was performed. After the leveling was completed, the nozzle returned to the origin. The MMU will then feed the filament into the nozzle, but the situation is like another picture. The MMU will continue to feed the filament to the nozzle. This process continues, but no melted filament flows out of the nozzle. I thought it was stuck somewhere, I pushed the filament manually, and the gears of the extruder turned extremely fast. Suddenly it seemed that the printing was completed and the screen displayed picture information. I want to know exactly where the problem is causing this printing to fail.（My expected effect is that I can use the screen function to load the filament and print the file from the SD card to run normally.）

vintagepc, 2020-01-15:
Your issues still sound as though they stem from misadjusted IR/FINDA sensors. As noted before, please check and adjust them.

catalincazacu86, 2020-04-09:
I think i solved my problem by checking all the cables at the mk3s board. 
The LCD cable was not good plugged in ( the one near the near where the door is closed ). 
I hope this will be helpful.

JohnnyDeer, 2021-02-03:
This is not a FW problem, please make sure your printer is properly assembled and calibrated. If this problem stay after checking, please contact our Support. I'm closing.

closed 2021-02-03 by JohnnyDeer