prusa3d/Prusa-Firmware #2415: LCD option to unload new filament on M600

Chompworks, 2020-01-14:
Problem:
When using M600, after using the LCD to remove filament, the user is prompted to load new filament. When that filament triggers the sensor, a small length is extruded and the LCD prompts for extruding more, resuming the print and a few other options. At the moment there is no way to remove the filament again in the current LCD menu flow. 

Proposed/Ideal solution:
Add a "unload filament" option (for non-MMU firmwares) AFTER the loading of new filament. In the below scenario, the user could then select this, and the firmware would invoke the unload procedure again, and then request the user to insert new filament again.

Why this is useful:
There is no easy way to remove a piece of filament if it was fed in with an unsuitable plug at the end. In that situation the filament plug tends to get stuck in the small bowden tube directly below the bondtech gears, but there is no LCD menu option currently to retract the filament. The only other option of forcing further extrusion is undesirable as it may grind the filament, and even break the plug off from the filament feed, necessitating opening up the extruder and removing it manually. Often that means there is no good way to save the print in progress.

If there was a way in the LCD menus to allow a user to choose to retract the filament, this would allow the user to reinsert the filament after trimming a problematic plug. The above scenario would be avoided, without loss of the print in progress.

wavexx, 2020-01-15:
I know I had this issues a few times (I opened the extruder to disengage the gears and pulled manually).

thew1se, 2023-05-13:
I was somehow able to get to the move axis menu and retract z-axis. And then resume print.  Can't remember how though...it's been a while.

Prusa-Support, 2023-07-06:
For the time being, the recommended method to get around these accidental occurrences is:
- open the idler door,
- guide the filament with needle pliers,
- purge more after the intervention.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-12-30:
Yes 🙂 by now it is merged too, hence ready for the next FW release.

Michele Moramarco
Prusa Research

closed 2024-03-11 by github-actions[bot]