prusa3d/Prusa-Firmware #2417: [BUG] Computing printer time in stats

montycz, 2020-01-14:
MK3S, 3.8.1

Computed printer time in stats compute also stopped prints. 

I builded and start printer on 25.12.2019. 
15.1.2020 I have more days, then possible.... Over month. I had much stopped prints during testing.

leptun, 2020-01-15:
Do you happen to have a MMU2(S) installed?

montycz, 2020-01-15:
No, standard MK3S

vintagepc, 2020-01-19:
IIRC there's a known issue where anytime a print pauses (MMU or not, it's definitely happening when the MMU hangs) the stats get inflated/bad.

Have you had mid-print filament runouts or crash detects?

montycz, 2020-01-19:
Ne, jen pokud vidím špatnout první vrstvu, špatný tisk, tak to stopnu to přes menu

montycz, 2020-01-19:
No, only if I see a bad first layer, a bad print, I'll stop it through the menu

leptun, 2020-01-19:
@vintagepc The pause time issue was MMU-related only. When you recovered from a load/unload error, the printer would have used the pause time to compensate for the time wasted, but that feature was broken for the MMU since it never saved the moment when it started the pause. This has been fixed in the latest 3.9.0-Beta.
However, you are saying that you don't have a MMU, so there might be something else...

vintagepc, 2020-01-19:
> @vintagepc The pause time issue was MMU-related only. When you recovered from a load/unload error, the printer would have used the pause time to compensate for the time wasted, but that feature was broken for the MMU since it never saved the moment when it started the pause. This has been fixed in the latest 3.9.0-Beta.
> However, he is saying that he doesn't have a MMU, so there might be something else...

I swear I've seen other comments from folks without MMUs, so there is probably a similar corner case somewhere. I did search and for the most part github search is failing me - but I did find some (old) mentions where the printer does not have an MMU:
https://github.com/prusa3d/Prusa-Firmware/issues/498

montycz, 2020-01-19:
I have no MMU.... I wrote it.

I tested it now.

Before start I have: 33d, 13h, 55m
I started print, stopped print
In statistic is 33d, 19h, 14m. 

the printing time was supposed to be 5:35...

I built a printer on 25.12.2019. Today is 20.1.2020. I have "printed" almost 34 days. It is impossible.

michalxfanta, 2020-01-20:
We'll fix it.

closed 2023-10-16 by github-actions[bot]