prusa3d/Prusa-Firmware #2419: Don't keep Bed heated if last unload failed

redlukas, 2020-01-16:
Hardware: i3 MK3S & MMU2S
Firmware: MK3S: 3.8.1, MMU2S 1.0.6
Observed Behaviour:
If the last unload of the Filament fails, the Printer goes into "MMU needs attention" mode. This lets the Nozzle cool down, but keeps the Bed warm
Expected Behaviour:
The printer checks if the failed unload operation was the last one. In this case, it would still display the "MMU Needs Attention" error, but let the heated bed cool down
Advantages if implemented:

- Less danger of the hot bed (fire/burn risk)
- Less power consumption
- If you are lazy like me and use a short bowden tube and a small piece of filament to trick the FINDA when printing in MMU Single mode to avoid the tedious process of loading the filament trough the Buffer the plate will stay hot, even if the print is otherwise finished and this will let the bed cool down.
[image]

wpegden, 2020-01-29:
Maybe it makes more sense for this to be a change in the slicer/gcode?  You just want to unset the bed temperature before the final unload instead of after it.  Logically, if you don't ask the firmware to change the temperature until after the unload, it seems like the correct behavior should be to wait until the load has succeeded.

ANTALIFE, 2020-08-09:
I suspect the reason for keeping the heatbed on is to keep the part secured to the surface. If it turns off then the thing you are printing will cool enough to naturally shrink and come off the bed

redlukas, 2020-08-09:
> I suspect the reason for keeping the heatbed on is to keep the part secured to the surface. If it turns off then the thing you are printing will cool enough to naturally shrink and come off the bed

how would that be relevant on the last unload? the print would be finished. I get why the bed is being kept heated on other filament change fails, but this issue is specifically for the last unload.

ANTALIFE, 2020-08-09:
Ah my bad, missed the "last" in unload

3d-gussner, 2020-08-11:
How should the printer know that this was the last unload command?

The printer executes one g-code line after another so if there is a user attention needed it will wait until the user does something or runs in an timeout. The bed temp is hold as @ANTALIFE mentioned to not loose the print.

The processor doesn't have the resources to load all g-code and analyse it.

closed 2023-10-16 by github-actions[bot]