prusa3d/Prusa-Firmware #2422: [BUG] During MMU2S print, if load fails bed temperature resets and extruder crash into bed

albegouws, 2020-01-17:
MK3S Firmware 3.9.0 Beta
MMUS2 Firmware 1.0.6

When load or unload fails, when the screen displays MMU load failed the set bed temperature goes back to zero.  In addition the extruder moves to the back left of the bed but then it moves down into the bed.  

If you press the buttons to resume the print heating starts and when set temperature reached the extruder stays on Z zero level.  As the bed temperature dropped the print is loose from the bed anyway.

Going back to 3.8.1 and 1.0.6 resolves the problem.

michalxfanta, 2020-01-20:
We'll try and fix it.

michalxfanta, 2020-01-30:
We are unable to replicate this issue.

DRracer, 2020-02-03:
@albegouws even though we were unable to replicate the issue as described, there have been multiple pause/resume issues fixed in the firmware (which is closely related to filament change) for FW 3.9 RC1. When the 3.9 RC1 goes out, please retest your scenario if it still happens.

JohnnyDeer, 2020-11-25:
Hi @albegouws is this issue still actual? Thank you.

JohnnyDeer, 2021-02-03:
I´m closing as "stale" and can´t reproduce.

closed 2021-02-03 by JohnnyDeer