prusa3d/Prusa-Firmware #2424: Custom sheet name is not displayed in home & setting menu

jsilva-iii, 2020-01-18:
MK2.5S + MMU2S
FW version : 3.8.1, 3.9 Beta

The name of a custom named sheet ( settings->HW-> HW Setup, Steel Sheets , rename a sheet) 
is not displayed on the home screen (in FW 3.9 Beta) or in the menu to change the sheet i.e. push selector button, last row, push button (to change sheet)
[image]
Home Screen showing sheet name Smooth1
[image]
Uner setting Steel Sheets, the renamed sheet

vintagepc, 2020-01-19:
Does it update on the home screen after a power cycle?

jsilva-iii, 2020-01-19:
No, I made the cahnge in FW 3.8.1, I power cycled a few times, then I updated to FW 3.9b and still same result.

vintagepc, 2020-01-19:
You may need to clear your data then - IIRC there was a slightly different layout for the first implementation of sheet names which might be causing you grief.

I just tried and cannot reproduce it; I renamed my sheet and the updated name showed instantly on the info screen. 

Additional thought: Have you scrolled down in your sheet list? There may be a second "smooth 1" and you renamed the one that's not selected. This could have happened when the number of sheets was expanded beyond the initial three or four.

jsilva-iii, 2020-01-19:
Ok, so I looked at the sheets available, I have 
HW Setup:
Sheet Smooth 1
Sheet Smooth 2
Sheet Texture 1
Sheet Texture 2
Sheet Custom  1
Sheet Smooth 1
Sheet Smooth 2
Sheet Texture 1
You are right, there duplicate names - how do I fix that, the renaming occurred in FW 3.8.1

vintagepc, 2020-01-19:
You have a few options:

1. Rename manually
2. Factory reset to reset them all to default
3. Since you are comfortable with firmware mods, my PR to add and M850 code that lets you alter sheet properties via serial: https://github.com/prusa3d/Prusa-Firmware/pull/2201

jsilva-iii, 2020-01-19:
Ok, renamed and fixed.. Thanks for the help.

closed 2020-01-19 by jsilva-iii

vintagepc, 2020-01-19:
Glad I could help! :)