prusa3d/Prusa-Firmware #2425: [FEATURE REQUEST]Percentage % instead of PWM 255

sarf2k4, 2020-01-19:
Sorry if this has been requested before, I'm not sure what this thing should be called.

I am requesting that the print fan value should be in percentage value 0-100% instead of pwm 0-255. 

Reason being is that it can be hard and forgetful to convert between 8bit value to percentage, in slicer, we even put them as 0-100% not 0-255. 

In the print fan value, I think there should be a small % symbol after the value showing that it is indeed in percentage value. Also maybe would be good if there's an option to switch between 8bit pwm and percentage value when adjusting them in the settings where 8bit pwm vs percentage preference should be kept even after power off.

michalxfanta, 2020-01-27:
Thank you for your idea, we will consider implementing.

jltx1, 2020-05-09:
I also vote for this enhancement.

closed 2023-10-16 by github-actions[bot]