prusa3d/Prusa-Firmware #2426: [BUG]Z height shows question mark during print

vintagepc, 2020-01-20:
**Printer type** - MK3S w/Skelestruder, 0.9 degree steppers on X/Y. 
**Printer firmware version**-  3.9.0-beta

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**Describe the bug**
While printing, there is a question mark after the Z height, even though the printer has run MBL as part of the startup sequence - and it should know the Z height during a print.
  
**To Reproduce**
Powered on, loaded MMU filament to nozzle, and started the attached print.

**Expected behavior**
Z should not be "unknown" during a print after MBL.
  
**G-code**
[Brick_test.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/4087512/Brick_test.gcode.zip)

**Picture**
  Please attach a video. It usually helps to solve the problem.
[image]

leptun, 2020-01-20:
There are some problems at the moment with this on the MK3S because of the PSU_DELTA define. Can you try disabling this define and compiling the fw and confirm if the issue persists?
I knew that it manifested only after the print, apparently there is more to it.
I’m not near a printer, so can’t confirm anything... :(

vintagepc, 2020-01-20:
Interesting - do you know the technical reason that define is implemented for the delta PSU? Curious the power supply affects Z handling.

I'll try the same print again to make sure it's not a fluke and then retry with the define removed.

leptun, 2020-01-20:
The PSU_DELTA define is meant to make the Z motors of the MK3S more silent. A side effect of the new PSU is that the always-on Z drivers were making quite some noise while idle. The solution to this was keeping them in stealthChop while they were technically unused.
This works quite well for silencing the steppers, but the implementation in code is quite messy tbh and it tends to fail in subtle ways.
The '?' reappearance after the stepper timeout (30s I think) where they are disabled (switched to silent in the case of Z) or when M84 is processed is caused by the last line in `disable_force_z()`. The usage of `axis_known_position[Z_AXIS]=false;` in this context is quite unknown to me since it makes no sense to mark it as unknown since the actual Z never moves. I think this is a relic of the past when the switching to silent of the Z motors would actually also unshift the Z axis to the closest fullstep just like in PowerPanic, but that idea was scrapped and I think that statement was just forgotten there.
In any case, I'm planning to implement an improved PSU_DELTA implementation that will not require the atmega2560 to deal with the switching to silent. I was thinking of using some neat functions of the TMC driver itself to switch automatically between modes when the firmware permits it.
You can clearly hear the difference here: https://photos.app.goo.gl/BEtata3YhxDmX9UR8
It switches modes when it starts and stops moving, so the transition is seamless.
You can also see here some very interesting planner bug that happens when the acceleration is really low (1mm/s/s) https://photos.app.goo.gl/49xVvie8ayHCWpub6
Btw, you have to listen at full volume to actually hear the stepper whine while idle (or it's absence in this case).
There is an easy way for you to hear the difference between the two modes. Just manually move the Z axis (the noise will be present), then disable the steppers (the noise will be gone). Keep in mind I'm only talking about Z noise, the others are disabled in the meantime.

leptun, 2020-01-20:
> I'll try the same print again to make sure it's not a fluke and then retry with the define removed.

@vintagepc I think you found the issue by chance. I wouldn't try too hard to reproduce it since it's most likely an edge case scenario regarding the MMU (probably). I'll get to polishing this feature later on in the 3.9 release. Right now, the PowerPanic "static memory has been overwritten" and spewing filament before crash is the main priority. Don't get me wrong, this will not be overlooked, it's just that this is the biggest issue with 3.9.0-Beta so far and it needs the most attention so that it can be right for the 3.9.0-RC1.
Take a look at 4:08 here: https://photos.app.goo.gl/3Wb6duTHssrvTXQA6

vintagepc, 2020-01-20:
Thanks for the added info, that was a really cool read!

I normally run in silent mode, so what could be happening is the MBL stealth->normal mode switching for the probing could be hitting the "clear known z" flag you mentioned.

No worries on the priority front, I know aesthetic bugs are not high up the list and only get worked on when all that remains to do is polish. 

Funny that the static memory overwritten is showing up elsewhere too - I know you couldn't repro #2140 but I actually got the same message there when I was going through my open bugs to see if any had been fixed in the beta. I wouldn't be surprised if there's a similar cause deep in some common function... planner corruption seems like a possible candidate.

leptun, 2020-01-20:
@vintagepc Well, in the case of this PowerPanic issue, it seems that the current_position for the Z axis was not properly initialized if the E axis was in relative mode (as it is now) and the first command would also move the extruder the entire part worth of plastic. Right now I'm waiting for confirmation from the others that the issue is really fixed. If it is, then I'll make a PR for 3.9.0-RC1.
Most "Static memory..." errors are caused either by lack of ram (stack too big, probably recursive calls to some processing function, usually lcd related or process_commands related) or by trying to write to an invalid pointer.
If you have spare time, would you mind giving https://github.com/leptun/Prusa-Firmware/tree/MK3-3.9.0_fix_PP a try and see if PP also works for you? If yes, then I suggest you try to see if you can trigger it in the first place and then test the fix. I suggest triggering it during the second layer on an infill section for a higher chance of getting the error.

leptun, 2020-01-20:
@vintagepc Can you elaborate on the second way to trigger the "static memory overwritten" error? I'm not sure what is the link to #2140

vintagepc, 2020-01-20:
Oops, swapped numbers. Meant to link #2410

leptun, 2020-01-20:
@vintagepc Ouf. That one... I still wasn’t able to generate the issue this way. Sorry.

vintagepc, 2020-01-20:
Aha! I watched it when starting the next print. The question mark does indeed appear about 30 seconds after MBL (I have a two-part preheat routine that does MBL at a lower nozzle temp to avoid ooze blobs). While waiting for nozzle/bed to hit final temps, there's a soft "thunk" noise, probably from the Z motors, and the "?" appears.

leptun, 2020-01-20:
Hmmm. It should never transition during a print. This means that the steppers (X,Y,E) are disabled and it might affect the positional accuracy in a negative way. I suggest you increase the inactivity timeout before the second temperature setpoint. Eg: “M84 S60” for 1min before steppers automatically disable.
I hope all your questions were answered regarding PSU_DELTA.

vintagepc, 2020-01-20:
Yeah, thanks, it satisfied my curiosity what was going on with that. Interesting on the transition - sounds like we might unintentionally be considering the M-commands to wait for temps idle time...

Could also explain what I thought was somewhat inconsistent first layers, if we're losing positional accuracy while waiting for things to finish heating...

leptun, 2020-01-20:
This only happens because the temperature wait commands don’t actively keep the steppers enabled. Normally this doesn’t happen, but you have a special start gcode.
Just thought of an interesting edge case: what if the steppers get disabled while the PowerPanic is heating for recovery? I’ll check this tomorrow when I have time.

vintagepc, 2020-01-20:
Hah. :laughing: Interesting how a simple "?" becomes a rabbit hole...

vintagepc, 2020-01-22:
Just ran a print with a boosted M84 S value, the ? has not appeared, so that is indeed the cause.

ochm, 2020-01-22:
I can confirm, when the steppers are turned off (after timeout or from menu), the question mark appears. Even if the Z motors are never really turned off (they are always locked).

closed 2023-05-13 by leptun