prusa3d/Prusa-Firmware #2428: [ENHANCEMENT]<Add fan speed to display>

Heckw, 2020-01-21:
This enhancement applies to: Printer type MK3S, MK3

I would suggest to add the fan speed to the display. On the 2nd line there should be enough space.
Since the fan speed changes often according to the slicer commands, it's hard to know if the fan speed correlates with a specific printing problem like stringing. Now you have to enter the menu several layers deep to get to the adjustments where you can read the fan speed. After some seconds the display returns to the info screen again. So it's really hard to watch the fan speed. On the other hand it could be so easy if the info display would only show the fan speed........

michalxfanta, 2020-01-27:
Thank you for your idea, we will consider implementing.

FrHePo, 2023-10-09:
For me this enhancement is not stale. I'm waiting for the implementation. The arguments have not changed.

3d-gussner, 2023-10-11:
@FrHePo Hello and thanks for the request.
At this moment we  can't change the Info screen layout without big UI change.
```
 |01234567890123456789
0|H215/215°   Z210.15?
1|B105/105°   >100%
2|USB100%     c--:--R?
3|Status line
```

3d-gussner, 2023-10-20:
Closing issue, sorry.

closed 2023-10-20 by 3d-gussner