prusa3d/Prusa-Firmware #2429: [BUG] Backward movement of extruder in intro line

ochm, 2020-01-21:
**Printer type** - MK3S
**Printer firmware version**-  3.9.0-beta-3175
**Describe the bug**
I found different behavior of into line extrusion between firmware 3.8.1 and 3.9.0-beta-3175.
In 3.8.1 firmware was everything correct. In 3.9.0-beta-3175 is the second part of inro line unprinted and the extruder stepper is moving backward.
  
**To Reproduce**
Use this start code:
M115 U3.8.1 ; tell printer latest fw version
M83  ; extruder relative mode
G90 ;absolute positioning
M140 S[first_layer_bed_temperature] ; set bed temp
G28 W ; home all without mesh bed level
G1 Z3
G1 X100 Y100 F10000 ;move extruder to center for better PINDA heating
M109 S160 ; Set extruder temp before bed level
M190 S[first_layer_bed_temperature] ; wait for bed temp
M117 Wait for PINDA temp
M860 S38 ; wait for PINDA temp
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
M109 S[first_layer_temperature] ; wait for extruder temp
M221 S100 ; Set flow to 100
M300 S440 P5
G92 E0.0
G1 X50.0 E10.0  F800.0 ; intro thin line
G1 X100.0 E6.0  F800.0 ; intro thiner line
G92 E0.0
M221 S{if layer_height<0.075}100{else}95{endif}

**Expected behavior**
Whe i moved M83 right before extrusion of intro line, the behavior is the same like with gcode above and firmware 3.8.1

M115 U3.8.1 ; tell printer latest fw version
G90 ;absolute positioning
M140 S[first_layer_bed_temperature] ; set bed temp
G28 W ; home all without mesh bed level
G1 Z3
G1 X100 Y100 F10000 ;move extruder to center for better PINDA heating
M109 S160 ; Set extruder temp before bed level
M190 S[first_layer_bed_temperature] ; wait for bed temp
M117 Wait for PINDA temp
M860 S38 ; wait for PINDA temp
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
M109 S[first_layer_temperature] ; wait for extruder temp
M221 S100 ; Set flow to 100
M300 S440 P5
M83  ; extruder relative mode
G92 E0.0
G1 X50.0 E10.0  F800.0 ; intro thin line
G1 X100.0 E6.0  F800.0 ; intro thiner line
G92 E0.0
M221 S{if layer_height<0.075}100{else}95{endif}

koenkooi, 2020-01-23:
I've seen a similar issue on my MK3S: the extruder will move backwards while loading filament through the MMU2S. With PLA you get a load grinding noise till the sensor triggers. With flex filament you don't get a grinding noise but a nice session of disassembling the extruder to get the spaghetti unwound from the gears.

Jamin82, 2020-01-23:
Can confirm extrusion changes to absolute mode during printer homing. Adding M83 to gcode after all homing is done results in extruder working again as expected

wavexx, 2020-01-27:
This should be fixed with #2416 (next beta)

closed 2020-09-21 by DRracer