prusa3d/Prusa-Firmware #2435: Changing PINDA position on firmware not working?

risingsunsun, 2020-01-24:
I'm currently designing a custom extruder with the PINDA positioned differently than the original extruder. I tried changing X_PROBE_OFFSET_FROM_EXTRUDER, but it doesn't seem to work.

During XYZ calibration at the "looking for bed sensor" phase (right after Z calibration), it moves to the correct position initially (i.e. where the new extruder's PINDA position is) then it moves right back to the old position (the original PINDA position) and makes that circular motion where it looks for the bed sensor.

Does anyone have any experience on this? thanks

madmaxed, 2020-04-10:
Same here.  I'm having the exact same problem.  Tried to use the mount linked for the Hemera extruder.  Updating firmware offsets do change how the printer reacts, but it isn't right.

I also designed up a new mount making the offsets much smaller than the Hemera mount already out there.  Similar result.

I'm digging through the code now and see tons of constants in addition to the OFFSET variables.

I see you posted this in January and still no response, hopefully we might get some others to chime in here.

risingsunsun, 2020-04-10:
@madmaxed I changed this constant and it worked. Can't remember which file this was in, but you can search the source

`const int16_t xyzcal_point_xcoords[4] PROGMEM = {1200, 22000, 22000, 1200};
const int16_t xyzcal_point_ycoords[4] PROGMEM = {600, 600, 19800, 19800};`

madmaxed, 2020-04-10:
Thanks, did you have to update the constants, and the X & Y PROBE offsets?

I'm not quite sure what these PROGMEM values are telling me?

ManuGithubSteam, 2020-04-19:
Hi Guys,
Im  nearly in the same boat. I did the vocano mod and steel springs under my bed so the height of the bed is screwd and not the ones the firmware is looking for when it tries to find the bed while printing (calibrations and stuff work fine tho....)

Does anyone know what value needs to change to let the pinda seach more than 1mm for the bed from a constant height ? I would like to have like 5mm or 10 at my disposal.

Thank you! and good luck. Lokking into the hemrea myself :-)

jrmontoyam, 2023-08-16:
Hi guys, 

I got the same problem. I  have designed my own exruder, which has the PINDA in a different position.  Did anyone get to solve it? It's beeing long time so I guess someone has a solution :)

closed 2023-10-24 by github-actions[bot]