prusa3d/Prusa-Firmware #2444: [BUG] Filament run out causes blank screen followed by garbage text on screen

kazibole, 2020-01-28:
**Printer type** - MK3S
**Printer firmware version** - 3.9.0-beta

**MMU Upgrade** - No

**Describe the bug**
During the filament run out process (unload, confirm unload, load, confirm colour), the screen will go blank and eventually display garbage characters. Pressing the knob still works to go through the filament run out steps. Once the new filament is loaded and the print resumed, the screen will return to the normal printing status.

I have noticed this happen multiple times and on previous firmware versions as well. I will see if I can reproduce it and provide more information.
  
**To Reproduce**
A print is started from the printer's SD card. OctoPrint is connected via USB to monitor the print. When the filament runs out the printer starts the filament change process. The screen will ask to press the knob to unload. I walk away for a few minutes and come back to find the screen blank. I can still press the knob to unload the filament. At some point the screen will display garbage characters. I can still press the knob to confirm that the filament was unloaded, load the new filament and confirm the colour. Once the print resumes, the screen returns to the normal printing status.
  
**Expected behavior**
The screen does not go blank nor display garbage characters.
  
**G-code**
N/A

**Video**
N/A - will try to reproduce

DRracer, 2020-01-31:
@kazibole are you sure this problem isn't caused by excessive static electricity - which is a usual cause of the "garbage on the LCD". There are screens which do not actively refresh themselves yet, thus the user may experience blanked LCD or garbage due to errors in communication between the CPU and LCD.
Please report if you can consistently reproduce the issue or if it doesn't happen again.

kazibole, 2020-01-31:
@DRracer Thank you for your response. It very well could be static electricity - the only time I touch the printer while its working is during a filament run out so that could have led to a false conclusion. I will continue trying to reproduce it and let you know.

Is there anything one can do to mitigate static causing this other than discharging any built up static before touching the printer?

DRracer, 2020-01-31:
@kazibole you can touch something conductive before touching the printer to get rid of excessive electric charge. 
We are planning to improve the LCD updates for FW 3.10, then it still may flicker in case of electric shock, but it shall repaint in a moment afterwards.

kazibole, 2020-02-03:
Unable to reproduce.

closed 2020-02-03 by kazibole