prusa3d/Prusa-Firmware #2445: [ENHANCEMENT] collect detailed filament loading statistics (MMU2S)

wpegden, 2020-01-28:
For the MK3S+MMU2S, it would be helpful if detailed statitics could be logged and checked regarding filament loading and unloading (not just when manual intervention was required).  For example, how often was second loading attempt required?

This would help dial in precise settings for very reliable operation, since the user would have more data to respond to other than total loading failures.

wpegden, 2020-01-29:
Just to be clear, the idea for these statistics is that they would be collected on a per-filament basis.  So for each filament, one could see how often loading slow moves were required after the initial fast move, how often a completely new loading attempt was made, and how often manual intervention was required.

michalxfanta, 2020-02-03:
Thank you for an interesting idea, this data would certainly help to identify some printing problems.

closed 2023-10-16 by github-actions[bot]