prusa3d/Prusa-Firmware #2456: [BUG] Acceleration/Jerk limits do not survive a power panic

wavexx, 2020-02-02:
**Printer type** - MK2.5/MK2.5S/MK3/MK3S
**Printer firmware version**-  3.9.0+

**Describe the bug**
Acceleration and Jerk limits are generally set at the beginning of a print and are usually different to default settings (or eeprom values).

These values currently do not survive a power panic, but they should.
On restart default values are reloaded, which will result in different printing behavior.

3d-gussner, 2023-11-07:
@gudnimg Aren't these now also stored in eeprom? Please verify.

gudnimg, 2023-11-08:
We definately don’t save the jerk value. Not sure about acceleration. It is very easy to add this. 👍

gudnimg, 2023-11-12:
It looks like we do save the acceleration. But we don't save the maximum jerk value.

`max_jerk` will require 4*4 = 16 bytes in EEPROM

We also don't save these values:

min_segment_time_us (4 bytes)

3d-gussner, 2024-04-09:
Fixed with #4657

closed 2024-04-09 by 3d-gussner