prusa3d/Prusa-Firmware #2462: [BUG] Bed does not stay heated on pause print

Photogad, 2020-02-05:
**Printer type** - MK3
**Printer firmware version**-  3.9.0 BETA

**MMU Upgrade** - no

**Describe the bug**
On pause print, heatbed goes to 0 temperature and starts cooling, causing the print to become detached.
  
**To Reproduce**
Add a pause print command in Prusa Slicer using the + button in sliced view
  
**Expected behavior**
Bed should stay heated to prevent print from coming unstick. Also, maybe should add setting though to turn the bed off after 6 hour of pause (in case user forgets the print is paused and doesn't want to burn down their house)
  
**G-code**

[DuoTabTower_0.15L_0.4N_PET.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/4156870/DuoTabTower_0.15L_0.4N_PET.gcode.zip)

wavexx, 2020-02-07:
I ran the print on a current build of the FW, and the pause keeps the bed on, so we probably have fixed this already.

I didn't test the full print on the  3.9.0 BETA, but on a smaller print with a pause command it also worked as expected (the bed was kept on).

vintagepc, 2020-03-05:
I also have not observed this with the current RC; bed remained heated when I paused a print or had "MMU needs attention"

Photogad, 2020-03-21:
> 
> 
> I also have not observed this with the current RC; bed remained heated when I paused a print or had "MMU needs attention"

mine is still cooling down on the 3.9.0 RC1 when the print pauses for any reason

Photogad, 2020-03-24:
@wavexx 
@vintagepc 

Have you tried keeping the print paused for more than a minute? It does not no always initially turn off the bed heat while the print is paused but it does after a minute or so sometimes. 

Here's a video of me pausing a print and it turning the bed heat to 0c within 10 seconds:
https://solasstudios.box.com/s/g6dk7xx3bnth9xs6mbfupc1775a7colg

You can see it sets the bed heat to 0 and the bed temperature slowly starts dropping (it will keep dropping until bed is cool). Even when I resume the print, you see it does not heat the bed back up, it stays at 0.

(I am on latest firmware release, 3.9.0 RC1)

vintagepc, 2020-03-24:
In my case I'm reasonably certain I was fiddling with the filament on my MMU for over a minute. It had run out and jammed while trying to eject the remaining scrap.  This was followed by probably another minute to install the new spool and feed it back up to the MMU.

Photogad, 2020-03-24:
> 
> 
> In my case I'm reasonably certain I was fiddling with the filament on my MMU for over a minute. It had run out and jammed while trying to eject the remaining scrap. This was followed by probably another minute to install the new spool and feed it back up to the MMU.

@vintagepc   maybe it works for you because you have MMU? I don't have it I am on MK3 (non S).  I don't know if maybe there's a difference in the firmware between those two variants. Maybe this bug only exists for original MK3 (non-S). I'm not sure.

I actually just paused the print again but this time the bed stayed heated for at least 3 or 4 minutes before I resumed.

So it seems this problem is intermittent and isn't always there. I wonder what changed between my last pause and this one, where the last pause didn't keep the bed heated but this one did?

I would say, from my experience so far the bed does not stay heated on about 65% of pauses. I will pause it a couple more times before this print finished and let you know

vintagepc, 2020-03-24:
Firmware is identical, but of course the MMU enables a few additional things internally.

Photogad, 2020-03-24:
@vintagepc  I paused the print again, and again the bed stayed heated. The only thing I can think of is that this bug only happens when the print is at a certain % of completion.  Whenever I paused this exact print before and had the problem of the bed turning off, it was between 20-50% done.  Now it's working properly and keeping the bed heated between 60-90% done.

wavexx, 2020-04-03:
Looking at the code itself, I don't see many places where this could happen. We definitely don't turn off the heat intentionally, and the safety timer shouldn't trigger because we're printing.

This leads to me suspect that maybe, while doing some other stuff, the "paused" status is lost for a fraction of a second where the safety timer has a chance to kick-in.

This is the only way I could explain this.
@leptun maybe you have some other idea?

Photogad, 2020-04-03:
> Looking at the code itself, I don't see many places where this could happen. We definitely don't turn off the heat intentionally, and the safety timer shouldn't trigger because we're printing.
>
> This [...]

@wavexx alright thanks for updating me.  I had to pause a print yesterday and it did the same thing, turned off the heaters on it's own shortly after i paused it.  So not every print has this problem, but I'd say most do. 

Also, it's worth mentioning im using the original silver MK3 PSU from 2 years ago, im not sure if that matters.

wavexx, 2020-04-03:
Do you recall doing anything at all just after pausing it? Could be anything, even dumb.
Maybe just navigating to some specific menu?

Photogad, 2020-04-03:
> Do you recall doing anything at all just after pausing it? Could be anything, even dumb.
> Maybe just navigating to some specific menu?

@wavexx , nope it does it on its own, usually a few seconds after I click "pause print" and the extruder begins moving up. Sometimes it seems to take longer though, but usually its immediate.

leptun, 2020-04-03:
@Photogad Do you happen to use the Preheat menu or do anything related to filament handling before the print where it triggers the issue?

leptun, 2020-04-03:
@wavexx I don't think that this is related to the safety timer. If it was, then a full screen message would have been presented. I'm basing my speculations from the video at the beginning of this thread.

Photogad, 2020-04-03:
> 
> 
> @Photogad Do you happen to use the Preheat menu or do anything related to filament handling before the print where it triggers the issue?

@leptun 

Sometimes I do preheat to load or unload filament and also do the preheat "cooldown" after I get it loaded. But I have not written a log to see if there's any correlation to that and when the problem occurs.

I am rebuilding my printer now to add the  MMU2S and MK3S extruder so when I get it rebuilt, I will do some tests for you.

But who knows, maybe once I add the MMU2S and MK3S extruder the problem will go away.

leptun, 2020-04-04:
@Photogad Do you have anything connected to the usb when it fails?

leptun, 2020-04-04:
What was the last version where you were able to generate the issue?
There are multiple theories as to what happened in the video.
The first theory is that the block_buffer (used for stepper motion) overrun its bounds and overwrote the target temperature. Here are the memory addresses of these items. The only way to be able t see if values have been overwritten would be to check `axis_known_position`, which brings me to the second issue I can see.
```
                0x000000000080075c                block_buffer
                0x0000000000800e5c                sm4_calc_delay_cb
                0x0000000000800e5e                sm4_update_pos_cb
                0x0000000000800e60                sm4_stop_cb
                0x0000000000800e62                _ZL13babystepLoadZ.lto_priv.321
                0x0000000000800e64                axis_known_position
                0x0000000000800e67                target_temperature_bed
```

I watched the video more carefully and noticed something unexpected: The time remaining and percent done are both cleared or revert to their usual states during a normal gcode print (not the case here). This should NEVER happen during a print.
The only way for these two to be cleared is by either being overwritten with 0xFFFF (unlikely) or having a M84 command get executed. This would also explain why I can hear the stepper motors disengage as soon as the pause movements are finished (`axis_known_position` will be set to 0)
Unfortunately, this is where the lead ends. I hope I could get some more information to hopefully identify what goes wrong in the FW.

I'd still like to know if octoprint or any other host is connected over USB to the printer. Should it be the case here, then we'd have another data point.

leptun, 2020-04-04:
Just for reference, it might be Related to #2422

Photogad, 2020-04-04:
> 
> 
> What was the last version where you were able to generate the issue?
> There are multiple theories as to what happened in the video.
> The first theory is that the block_buffer (used for stepper motion) overrun its bounds and overwrote the target temperature. Here are the memory addresses of these items. The only way to be able t see if values have been overwritten would be to check `axis_known_position`, which brings me to the second issue I can see.
> 
> ```
>                 0x000000000080075c                block_buffer
>                 0x0000000000800e5c                sm4_calc_delay_cb
>                 0x0000000000800e5e                sm4_update_pos_cb
>                 0x0000000000800e60                sm4_stop_cb
>                 0x0000000000800e62                _ZL13babystepLoadZ.lto_priv.321
>                 0x0000000000800e64                axis_known_position
>                 0x0000000000800e67                target_temperature_bed
> ```
> 
> I watched the video more carefully and noticed something unexpected: The time remaining and percent done are both cleared or revert to their usual states during a normal gcode print (not the case here). This should NEVER happen during a print.
> The only way for these two to be cleared is by either being overwritten with 0xFFFF (unlikely) or having a M84 command get executed. This would also explain why I can hear the stepper motors disengage as soon as the pause movements are finished (`axis_known_position` will be set to 0)
> Unfortunately, this is where the lead ends. I hope I could get some more information to hopefully identify what goes wrong in the FW.
> 
> I'd still like to know if octoprint or any other host is connected over USB to the printer. Should it be the case here, then we'd have another data point.
> [MEMORY_MAP_MK3-c8f67f22791b65840211d5616cb541100b40e6bb.txt](https://github.com/prusa3d/Prusa-Firmware/files/4432230/MEMORY_MAP_MK3-c8f67f22791b65840211d5616cb541100b40e6bb.txt)

@leptun I THINK YOU SOLVED IT!!

I DO have octoprint connected to the printer! I am NOT using octoprint to start the print, I am only using it to send me e-mails when the print finishes, and to take a live video feed of a webcam. I am still using the SD card and printer menu to start the gcode prints.

So I guess that's probably what is messing it up though, for some reason when the print is paused, the octoprint monitoring is doing something weird to the heater?

I'm not sure why though!

To be 100% sure, I will run some tests with the raspberri pi disconnected from the printer

leptun, 2020-04-04:
Please post the gcode scripts from your octoprint instance (Settings). All of them. Screenshots are ok.

Photogad, 2020-04-04:
> 
> 
> Please post the gcode scripts from your octoprint instance (Settings). All of them. Screenshots are ok.

@leptun 

it appears to have gcode to turn off the heaters when the print is canceled, but not when the print is paused.

Could it be that for some reason the mk3 firmware is making octoprint think a pause print is actually a cancellation? 

[image]

leptun, 2020-04-04:
There’s no way for me to know except testing and logs from octoprint. Both Terminal logs and the internal octoprint server logs are useful.

I’ll try to see if I can reproduce the issue on my test bench tomorrow with the latest octoprint and the default gcode scripts.

leptun, 2020-04-04:
What is that “E-Stop settings” plugin?

Photogad, 2020-04-04:
> 
> 
> There’s no way for me to know except testing and logs from octoprint. Both Terminal logs and the internal octoprint server logs are useful.
> 
> I’ll try to see if I can reproduce the issue on my test bench tomorrow with the latest octoprint and the default gcode scripts.

@leptun 
here's a log file from the day when I was running all the videos showing the pause print not keeping the bed heated.

[octoprint.log.2020-03-24.txt](https://github.com/prusa3d/Prusa-Firmware/files/4432393/octoprint.log.2020-03-24.txt)

It's a big one and makes no sense to me, but maybe you can decipher it quicker.

Photogad, 2020-04-04:
> 
> 
> What is that “E-Stop settings” plugin?

not sure. it says this

[image]

Photogad, 2020-04-04:
@leptun  here's a snippet from the octoprint log that seems to be what we are looking for?

`2020-03-24 21:26:01,322 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Operational"
2020-03-24 21:26:01,331 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Paused"
2020-03-24 21:26:39,503 - octoprint.util.comm - INFO - Externally triggered heatup detected
2020-03-24 21:27:13,580 - octoprint.util.comm - INFO - Timeout while in an active heatup, considering heatup to be over.
2020-03-24 21:27:14,611 - octoprint.util.comm - INFO - Changing monitoring state from "Paused" to "Resuming"
2020-03-24 21:27:14,647 - octoprint.util.comm - INFO - Changing monitoring state from "Resuming" to "Printing from SD"
2020-03-24 21:27:18,922 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2020-03-24 21:27:45,461 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2020-03-24 21:38:17,166 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-03-24 21:39:24,337 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2020-03-24 21:40:25,331 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2020-03-24 21:41:47,365 - octoprint.util.comm - INFO - Changing monitoring state from "Printing from SD" to "Pausing"
2020-03-24 21:41:47,441 - octoprint.util.comm - INFO - Changing monitoring state from "Pausing" to "Paused"
2020-03-24 21:42:09,058 - anywhere.mjpeg_stream - ERROR - [Errno 32] Broken pipe
2020-03-24 21:42:14,784 - octoprint.util.comm - INFO - Externally triggered heatup detected
2020-03-24 21:42:41,858 - octoprint.util.comm - INFO - Timeout while in an active heatup, considering heatup to be over.
2020-03-24 21:42:43,795 - octoprint.util.comm - INFO - Changing monitoring state from "Paused" to "Resuming"
2020-03-24 21:42:43,827 - octoprint.util.comm - INFO - Changing monitoring state from "Resuming" to "Printing from SD"`

I guess where it says `octoprint.util.comm - INFO - Externally triggered heatup detected` would be where I used the button on the printer to heat the bed back up after the printer automatically turned it off

leptun, 2020-04-05:
Hmmm. Do you also have one where the Serial communication can be seen? Or even a separate log?

Photogad, 2020-04-05:
> 
> 
> Hmmm. Do you also have one where the Serial communication can be seen? Or even a separate log?

@leptun  it doesn't appear so, or at least I can't find it.

FYI I finished my MK3->MK3MMU2S rebuild today and did a few prints with the octoprint disconnected from the printer. I was able to pause the print with no problems several times, bed stayed heated.

So it's narrowed down to:

1. OctoPrint was causing the bed to turn off.

or

2. Upgrading to MK3S firmware fixed it.

I will do some more tests when I can with the OctoPrint connected and see if the problem still remains.

Photogad, 2020-04-08:
Well, I've reconnected the OctoPrint, but now the bed always stays heated it seems during the pauses.

So my problem is solved. I'll re-open this if it comes back. If anyone else has these issues, tell them to upgrade their OctoPrint software or to upgrade to the MK3S firmware from the MK3 firmware. One of those fixed it.

closed 2020-04-08 by Photogad