prusa3d/Prusa-Firmware #2464: [BUG][3.9.0-BETA] Printer resets on MMU filament runout

vintagepc, 2020-02-05:
**Printer type** - MK3S
**Printer firmware version**-  [3.9.0-BETA]

**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version** - 1.0.6

**Describe the bug**
Was running in single mode near the end of a spool. The printer paused when the FINDA  detected a runout. It tried to unload but the tail  of the filament got caught. This caused the extruder to chew into the filament until the IR sensor also read 0. At this point the printer  rebooted. 

Possibly related/shared cause to #2301, #1993, but certainly more severe - the linked issues prevent a print from starting, but this situation can result in a failed print after many hours and much material.

**To Reproduce**
Simulate a filament runout followed by a situation where both sensors report 0. 
  
**Expected behavior**
Printer should pause and MMU should indicate it needs attention. 
  
**G-code**
N/A, can be done with any gcode.

**Video**

jurcja1, 2020-02-05:
I would like to add similar bug with beta 3.9.0 FW and MMU2s. 

When filament runs out, printer is waiting for new filament (message on display) to be feeded in MMU unit and then continuing in print, but MMU unit took filament back but does not park selector beside used filament and does not push old filament to take it out. Is anyone facing same issue during testing BETA FW with MMU2s? 

Thank you.

vintagepc, 2020-02-05:
> I would like to add similar bug with beta 3.9.0 FW and MMU2s.
>
> When filament runs out, printer is waiting for new filament (message on display) to be feeded in MMU unit and then continuing in print, [...]

I believe you're expected to remove the runout from the rear of the MMU, the reverse of what you'd do to load a new one.

jurcja1, 2020-02-05:
> 
> 
> > I would like to add similar bug with beta 3.9.0 FW and MMU2s.
> > When filament runs out, printer is waiting for new filament (message on display) to be feeded in MMU unit and then continuing in print, but MMU unit took filament back but does not park selector beside used filament and does not push old filament to take it out. Is anyone facing same issue during testing BETA FW with MMU2s?
> > Thank you.
> 
> I believe you're expected to remove the runout leftovers from the rear of the MMU, the reverse of what you'd do to load a new one.

I thought and that was in previous version, that when filament runs out printer is waiting and filament is in MMU unit like on the picture.

[image]

vintagepc, 2020-02-05:
I could very well be wrong.  👍

vintagepc, 2020-02-06:
Filed #2466 to address this. Both of my situations were caused by a filament runout while the current filament was in an unknown state. 
 The PR makes it handle this state gracefully  by requiring a user filament change.

PR #2466 (2020-03-26): #2464 fix runout for unknown filament (also fixes #1993, #2301)
This fixes the printer soft-resetting if a filament runout is detected while the current filament is unknown (e.g. user loaded to nozzle, and an event caused a power cycle (power panic during MMU print), or user turned off printer and continued printing with the same filament the next day). 

Because the current filament is unknown, we cannot reliably retract it and move on to the next one if spooljoin is enabled (theoretically we could try this and let the MMU try a retract/home/next filament, but that seems risky),  nor can we m [... cut]
closed 2020-03-26 by DRracer