prusa3d/Prusa-Firmware #2471: [ENHANCEMENT] Default nozzle temperature in Settings menu should be ~200

ax42, 2020-02-07:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - all

**Describe the solution you'd like**
Currently, if I want to manually set the nozzle temperature from the Settings menu (e.g. to unload filament without heating the bed), the initial value is zero and I need to turn the knob many times to get to temperatures around 200 degrees suitable for unloading.  I suggest that the initial value should be (say) 200, or that the initial value can be set in the settings menu.

ManuGithubSteam, 2020-04-19:
This is correct however some filaments need less Temp like PVA so i would say 150 degress is good but definitly more that 0...

Panayiotis-git, 2020-04-19:
@ax42 Check the https://github.com/prusa3d/Prusa-Firmware/pull/2335

Prusa-Support, 2023-08-23:
The relevant PR is #3494 and I'm glad to announce it was merged on August 2.

This issue is resolved but the merge occurred slightly before the FW release 3.13.0 so the firmware will probably officially see this implementation on release 3.13.1 (which is already in the works).

Michele Moramarco
Prusa Research

closed 2023-08-23 by Prusa-Support