prusa3d/Prusa-Firmware #2472: [ENHANCEMENT] Warn that stealth mode can't be used for belt calibration.

Strat00s, 2020-02-07:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MK3s

**Is your enhancement related to a problem? Please describe.**
I was trying to "calibrate" my belts via the belt status option inside calibration menu and every time, a printer told me that there was an error and that I should "check axis length x". But all I needed to do was to disable stealth mode.

P.S.: The calibration was kinda useless anyway, as even after some serious tightening, the belt status went up instead of down.
  
**Describe the solution you'd like**
Maybe add a warning that it can't be calibrated in stealth mode or automatically disable the stealth mode before the "calibration".

leptun, 2020-02-07:
@Strat00s Thanks for reporting.
We’ll fix this issue for the final 3.9.0 release.

vintagepc, 2020-02-07:
That should also fix #2356...

vintagepc, 2020-02-07:
I have a tentative fix in https://github.com/vintagepc/Prusa-Firmware/tree/%232472-%232161-belt-test-stealth-mode
Will test tonight and PR.

PR #2492 (2020-03-26): Belt test improvements (No display alteration)
Per @leptun's request, this PR contains only the the non-display-altering changes for the belt test fixes included in #2473 (i.e. fixes #2472 and fixes #2356 only)

No new translations were required for the display changes but we might want to do further cleanup of the display layout in this function before making that change. 

PFW-1102
PR #2473 (2020-04-20): #2472 #2356 #2484 belt test improvements
Forces high power mode for belt test.

- Fixes #2472, Fixes #2356, resolves #2484 

Edit: Also fixes #968 , closes #2547 

PFW-1104
closed 2020-03-26 by DRracer