prusa3d/Prusa-Firmware #2475: [ENHANCEMENT] Allow using SD slot extenders

lachs0r, 2020-02-08:
So I wanted to use a custom front panel that has the SD card slot in a different position. I thought I’d be able to use an extension cable, but the printer detects a card as soon as it is plugged in, whether or not it actually has a card inserted at the other end.

Of course, accessing a non-existent card fails, and the printer assumes there is no SD card until the extender is removed and inserted again. It works fine when the extender is inserted with a card in its slot.

This can also be reproduced with microSD adapters.

To solve this, maybe the printer could *always* attempt to communicate with the SD card when selecting the menu entry, even when the last attempt has failed. Or perhaps there are different ways to detect when a card is actually inserted?

leptun, 2020-02-08:
You can compile the firmware on your own. All you have to do is undefine the SD detect pin and you will have options to release/initialize/refresh SD card and it should work as expected in your case. SD card detection will not work at all in this mode, but it should still work.
I have to check whether that code still works as it is quite old, but in theory it should work. I’ll make a post with the required changes once I test them (on Monday).

lachs0r, 2020-02-08:
Thank you! I assume it’s this line: https://github.com/prusa3d/Prusa-Firmware/blob/c122399020224496e2acb5770f9c79d17ba891c1/Firmware/pins_Einsy_1_0.h#L117

leptun, 2020-02-08:
Yes, but I’m not sure if the rest of the code still works. You can test it, though. I’m actually really curious if it was mentained.

lachs0r, 2020-02-08:
I guess the actual problem is that the extender I bought is meant for microSD slots, which lack the additional unused pin, and I’m using this with a microSD-to-SD adapter. It’s pretty hard to find regular SD extenders these days, and I wasn’t thinking when I made that purchase.

I can probably do a slight modification to the adapter (additional wire) so it works as intended. But if the software mod works, that will suffice in the short term. Will try it this weekend.

vintagepc, 2020-02-18:
I don't think this can be implemented in software. the CD (card detect) pin is a physical switch in the socket, not an electrical presence detect. So you could insert a shaped piece of plastic and still have it think there is a card present. You'll need to rewire the CD switch from the main socket to your extender's CD switch.

closed 2023-10-19 by github-actions[bot]