prusa3d/Prusa-Firmware #2484: [ENHANCEMENT] Better "Belt test" implementation

leptun, 2020-02-11:
As of now, the function always raises the Z a certain amount. `raise_z_above()` should be used instead.
Also the LCD rendering could be improved.

Note for later.

vintagepc, 2020-02-11:
I'm happy to include this in my open PR on the function. 
Is there a suggested layout for the LCD rendering or is that more of a "could be better, not sure how yet"?

leptun, 2020-02-11:
The layout is mostly ok. I’m more concerned about the way the screen is rendered.

vintagepc, 2020-02-11:
Ah, the custom LCD_belttest_print function that redraws everything? I did think that could be cleaned up. Something more compact/efficient that only updates the relevant display section, correct?

leptun, 2020-02-11:
Yep

PR #2473 (2020-04-20): #2472 #2356 #2484 belt test improvements
Forces high power mode for belt test.

- Fixes #2472, Fixes #2356, resolves #2484 

Edit: Also fixes #968 , closes #2547 

PFW-1104
closed 2020-04-20 by DRracer