prusa3d/Prusa-Firmware #2485: [QUESTION] Building under linux

Thelvaen, 2020-02-11:
Hello,

I'm pretty sure "sudo" is not needed to compile the firmware (except for package installation), hence I believe doc should be edited to remove the "sudo" in front of the build.sh.

leptun, 2020-02-12:
@Thelvaen 
3d-gussner: I am not using build.sh often but I think it does work without sudo. Looking at the travis config you can see that it runs without special rights. All depends on the folder rights.
PF-build.sh usually runs without any special rights except if apps are missing the user gets a help message to use sudo apt-get install.. To install them. So the README.MD  can be updated I guess.
Does it work correctly in your case without sudo?

Thelvaen, 2020-02-12:
Yup, it did work just fine on my CentOS 7 installation, firmware has been compiled successfully and flashed to my printer without issue.

Hence, if it works for other people as well, the README.MD should be amended. As a sysadmin I tend to try and use the less privileges as possible, so if something goes wrong, it does not go really bad.

As I'm not a regular contributor here, I'd rather have someone from the project (or from Prusa) to submit this modification, but if needed, I'll do it.

leptun, 2020-02-12:
Perfect. Anyone is allowed to make changes and submit Pull Requests. You can do it here: https://github.com/prusa3d/Prusa-Firmware/edit/MK3/README.md
It would be better if you did it since I don't use linux and it would be difficult for me to test everything afterwards.

closed 2022-12-29 by wavexx