prusa3d/Prusa-Firmware #2486: Unable to flash firmware on Einsy Rambo 1.1b

papaiatis, 2020-02-11:
I'm trying to flash the most recent stock firmware (3.8.1) on my recently bought Einsy Rambo 1.1b. 

I plugged the machine to power outlet and connected the USB cable. Using Prusa Slicer I started to flash the firmware. The writing and reading process were successful but the third step fails.
Here's the output: https://pastebin.com/T2J0fy8Z

Any idea what could be the problem?
Thanks!!!

leptun, 2020-02-12:
Can you please post a photo with the front of the board? It fails to upload the language section of the hex file on the external chip. Maybe there is an issue with it.
In any case, if you don’t use languages other than english, this will not influence the functionality of the firmware for now.

leptun, 2020-02-12:
Also, from where did you buy the einsy board?

papaiatis, 2020-02-12:
Hi @leptun 
Thanks for your reply. English language is enough for me. So you are saying that I should be able to start the board/printer even though the third process failed?
Because I'm not able to. The LCD is blank too. 

Also I'm not sure if it has a valid bootloader or not. 

I bought it from China.
Let me know if you need more photos.

[image]

[image]

[image]

leptun, 2020-02-12:
Ok. This changes things a lot. You have to flash the custom bootloader on the atmega2560 and also set the correct fuse settings.
You need an icsp programmer for this or an arduino as isp.
I think that if you do that everything should work.

papaiatis, 2020-02-12:
I have some Arduinos around. 
Could you please point me to the place where I can download the mentioned custom bootloader? And the fuse settings?

leptun, 2020-02-12:
I’ll try to search for some good instructions for this.

leptun, 2020-02-12:
Ok. So I tried compiling a list of instructions:

- Open Arduino IDE:
- Select board as the arduino board you will use as a programmer (probably uno)
- Open the sketch in `File -> Examples -> Arduino ISP`
- If you use something else than UNO, change the pin definitions in the sketch
- Upload the sketch to the UNO (make sure you use the correct port (the UNO, not the printer board))
- Connect the boards as follows:
[image]
- In Arduino navigate `to File -> Preferences -> Settings`
- To the text field "Additional Boards Manager URLSs" add https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/package_prusa3d_index.json
- Open Board manager (Tools->Board->Board manager), and install `Prusa Research AVR Boards by Prusa Research` (very important!)
- Select `Tools -> Board -> PrusaResearch Einsy RAMBo`
- Make sure the port **STILL IS SET TO UNO!**
- Select the 'Tools -> Programmer -> Arduino as ISP'. (NOT ArduinoISP! made the mistake myself and took me a while to figure out why it wasn't working)
- Click `Tools -> Burn Bootloader`
- After the process finishes, unplug everything and disconnect the two boards.
Now you should be able to upload the firmware using the printer board's USB port using PrusaSlicer.

Let me know if this fixes your issue.

papaiatis, 2020-02-12:
Thanks a lot @leptun  for the great instructions!
I managed to upload the bootloader and after that I also managed to upload the firmware with PrusaSlicer.
However, the machine still not starting up. I still have a blank LCD.

I connected the printer to my laptop and checked the serial debugs and that's what I see:

[image]

"SD init fail" got my attention. I have an old (~5yrs) LCD display which worked fine for my previous printer. On the back of the LCD I see these texts: "RepRapDiscount Smart Controller" and "Robotale 3D_Print" and "2013-0729 SmatrConteallerD.PCB".

When I push the button I hear the click sound though.

leptun, 2020-02-12:
Maybe one of the wires is reversed? Otherwise I don’t know why it wouldn’t work.

papaiatis, 2020-02-12:
So what do you think about "SD init fail" ? Could that be the culprit why the LCD does not work too? The board can't communicate with the LCD?

leptun, 2020-02-12:
There seems to be one more issue with it: The W25X20CL (SPI external flash for languages) is not initialized successfully. Does it initialize correctly if you unplug the LCD cables (error message gone)?
If the cables are swapped/connected incorrectly in any way, the SPI bus might not work properly.

Here are some LCD schematics that might help you:
[image]
[image]

papaiatis, 2020-02-12:
When LCD is unplugged the line "W25X20CL not responding." is missing from the output.

Since this issue was about the flashing, I'm closing this now since it was resolved.
I opened another ticket for the LCD issue here: https://github.com/prusa3d/Prusa-Firmware/issues/2489

closed 2020-02-12 by papaiatis

husqy-701, 2022-02-15:
@leptun 
I'm trying to follow the steps that you wrote above and having an issue.
when im trying to choose the programmer as so:
 'Tools -> Programmer -> Arduino as ISP'.
it says "no programmer available for this board" (as shown in the picture)

also, i tried to use the second board "Original prusa i3 mk3 multi material 2.0 upgrade"  that came with it when i installed the prusa board as written in your instructions.
and stumbled upon error while burning bootloader "avrdude: Yikes!  Invalid device signature."
(also shown in picture)

Thanks for everything you wrote here!

leptun, 2022-02-15:
@husqy-701 Huh. That's really weird. I have the same issue as you. I am using arduino 1.8.16.
I just tested on arduino 1.8.5 and the programmers are back. For the time being, please use that old version. Here is a download link of arduino 1.8.5 for windows in zip file format (not installer): https://downloads.arduino.cc/arduino-1.8.5-windows.zip

@3d-gussner fyi. We should investigate what changed in arduino that broke our board definitions.

husqy-701, 2022-02-16:
@leptun 
It worked, the programmer did show up this time!
but im stumbling upon the same error as before while burning.

[image]
[image]

leptun, 2022-02-16:
It would help if you showed the entire log. What signature does it read back? If it's 0x000000 or 0xfffffff or something along those lines, then it's a wiring error, so check the wires and try again.

husqy-701, 2022-02-16:
Tried to find on the web how to read the signature. 
unfortunately i'm new to arduino and couldnt find how to do that.

could you explain please how to do that?

btw, checked the wiring again and all seems good.
maybe, its worth mentioning that when i assembled the printer i accidantly connected the lcd cables inverse and than corrected it.
after that, i ran the setup wizard until it had a issue with one of the axis.
turned off the printer, checked the axis. than tried to ran again and it stuck on "prusa reaserch original i3" for ever..

leptun, 2022-02-16:
[image]
I think you need to enable those to get any meaningful information besides the error.

husqy-701, 2022-02-16:
You right!

this is the full error message:

Arduino: 1.8.5 (Windows 10), Board: "PrusaResearch Einsy RAMBo"

C:\Users\ilan1\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\ilan1\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf -v -patmega2560 -cstk500v1 -PCOM9 -b19200 -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xD8:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\ilan1\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : COM9
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0xffffff (probably .avr8x_mega) (retrying)

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0xffffff (probably .avr8x_mega) (retrying)

Error while burning bootloader.
Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0xffffff (probably .avr8x_mega)
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

so, i Checked the wiring again...
No luck again...

maybe this board is dead....

[image]
[image]

leptun, 2022-02-16:
Do you get something on the LCD if you power up the board? If you do, then the existing bootloader code is functioning and the board is not dead

husqy-701, 2022-02-16:
yeah.
this what i get, and it stuck like that for ever.
[image]

leptun, 2022-02-16:
Then it's not fried. There is no "..." in the bottom right, so the firmware attempts to boot. Open a serial port to the einsy board via its usb port at baudrate 115200 and post what the board says during bootup

husqy-701, 2022-02-16:
when the board turned on it wont send anything.
and if i press ENTER few times it says "command not found!"
[image]

leptun, 2022-02-16:
What do you get back if you send `;C32u2_FWV`?

husqy-701, 2022-02-17:
im getting this
[image]