prusa3d/Prusa-Firmware #2487: [ENHANCEMENT] Power Panic Preheat Nozzle Before Moving

fromanza, 2020-02-12:
MK3s & MMU2s - latest FW as of 12/2/2020

Problem is when a power failure occurs and when the user selects to resume, the extruder head moves BEFORE the nozzle is heated which means all the oozed material that has come out during the power failure has cooled and stuck to the print and this often results in the part being unnecessarily dislodged from the bed failing the print anyway. Particularly with small footprint prints and especially when using the textured bed.

Simple solution, preheat the nozzle to at least say 180-200 degrees before moving after a PP.

leptun, 2020-02-12:
This issue will be fixed in the upcoming 3.9.0-RC1. The head moves away during power panic so no material should be stuck to the part.

gudnimg, 2023-07-15:
I think we're safe in closing this issue. Was probably fixed quite some time ago.

Power panic saves whatever the last "target temperature" was e.g. 215°C for PLA and preheat to that temperature.

closed 2023-07-15 by gudnimg