prusa3d/Prusa-Firmware #2489: [QUESTION] LCD is blank and I'm getting "W25X20CL not responding." in the console

papaiatis, 2020-02-12:
I'm about to finish building my MK3S printer. I had some troubles with flashing the firmware that I reported here: https://github.com/prusa3d/Prusa-Firmware/issues/2486
Now I (probably) have a working bootloader and firmware and wanted to start up the machine but the screen just not turning on at all. I checked the cables 100x times, tried them in reverse, swapped them to no avail. 
I'm seeing the following output in the console with two LCD related errors:

[image]

I went through all the reported issues here that mentions W25X20CL but neither of them helped.
I tried disconnecting all cables except power from PSU and LCD cables, didn't help.
I tried flashing earlier firmwares with PrusaSlicer and also tried flashing the recent one using Arduino IDE, didn't help.
I tried commenting out `#define W25X20CL` but then the buzzer just keeps beeping when I turn the printer on. 
I can measure 5V between pin 9 and 10 on EXP1 connector so probably LCD gets the power. 

I also have a spare Arduino Mega and RAMPS 1.4 on which I'm still running an old version of Marlin. I connected the LCD to this board, and connected the board via USB to my laptop and the LCD was working. 

I'm out of ideas now. 
Could someone please help?

leptun, 2020-02-12:
What happens if you only connect the P1 cable? In my case only the display works correctly (and click button). Try this so that we can really rule out the connector possibility.

I can only help you with small steps so that I can maybe figure out what is going wrong.
Also, is the LCD identical to the one on the Prusa? Maybe a picture of the back of the lcd pcb?

leptun, 2020-02-12:
I find it really weird that the cable swapping and reversing didn’t help in any way. The buzzer thingy is the way my printer behaved at some point when I wired my lcd the wrong way.

papaiatis, 2020-02-12:
Connecting only P1/EXP1 resulting in this output:

[image]

Still, the LCD is blank. Not even the backlight turns on.
If I swap EXP1 with EXP2 then the screen flickers for a couple of times but that's it.

leptun, 2020-02-12:
This still sounds really weird. At the very least with only P1 connected the SPI is no longer broken.
Which EXP2 connector did you connect? P1board-P2display or P2board-P1display?

papaiatis, 2020-02-12:
[image]
[image]

papaiatis, 2020-02-12:
> Which EXP2 connector did you connect? P1board-P2display or P2board-P1display?

Sorry, I'm not sure I follow this question. 
By default it should be EXP1 -> P1 and EXP2 -> P2, right? 

Indeed it is really weird. I'm thinking maybe there is still something wrong with the bootloader/firmware. Or I have a faulty Einsy board.

leptun, 2020-02-12:
Just pointing it out, the connector housings on the lcd module seem to be rotated 180deg compared to the official prusa lcd.
[image]

leptun, 2020-02-12:
> Sorry, I'm not sure I follow this question.
> By default it should be EXP1 -> P1 and EXP2 -> P2, right?
>
> Indeed it is really weird. I'm thinking maybe there is still something wrong with the [...]

You said you did some swapping, so I was asking which way you connected them just so that I was sure I understood the way you connected them.

papaiatis, 2020-02-12:
Okay, by swapping I meant EXP2 -> P1 and EXP1 -> P2. 

> Just pointing it out, the connector housings on the lcd module seem to be rotated 180deg compared to the official prusa lcd.

Well, that could be the indeed the problem. Unfortunately I can only plug the connectors in one way. So let me cut off that piece of plastic that prevents me to doing so.

leptun, 2020-02-12:
I think that you can carefully slide the plastic cover off and insert it the other way on the pcb. The metal pins should remain on the board.

papaiatis, 2020-02-12:
OMG!!! That was it!
I carefully removed the cover and plugged the cables 180degrees and it worked!
I own you a lot. I really appreciate your time and help!
I'll create new ribbon cables and will put the connector the other way around so I can put the covers back.

closed 2020-02-12 by papaiatis

leptun, 2020-02-12:
Glad I could help.