prusa3d/Prusa-Firmware #2490: Please lower BED_MINTEMP also for MK2.5 and MK2.5S

matzman666, 2020-02-12:
I have a MK2.5 and since some time I have the problem that whenever I start a print I get the error message "Temperature heated bed switched off. MINTEMP triggered!" after about a minute.

From what I could piece together the problems seems to be related that my bed cannot get warm enough in time. According to the code the bed temperature timer as an interval time of 50s and the min temp treshold is 30°C. I live in Western Europe where we currently have winter and therefore ambient temperature is lower than normal. This leads to my bed needing longer than 50s to reach 30°C (it's a close call, after 50s the bed temperature is usually around 29°C) triggering the MINTEMP error.

The MINTEMP for MK3 and MK3S apparently has already been lowered. Could this also be done for the MK2.5 and MK2.5S?

DRracer, 2020-02-13:
@matzman666 unfortunately this cannot be done reliably for the MK2.5 due to electrical properties of the measurement circuit. The MK2.5 is only able to sense at least 15°C.

Naugrimohtar, 2020-03-13:
This is a bummer! I have a MK2.5 I have in my basement, typically 55F (~12-13C). After I start a print when it is cold, it begins to warm up but then fails with MIN Temp. I try again and it works. This is a strong annoyance.

Naugrimohtar, 2020-03-17:
Here is my log
`Changing monitoring state from "Detecting baudrate" to "Operational"
Send: N0 M110 N0*125
Recv: echo: 3.9.0-RC1-3272
Recv: echo: Last Updated: Feb 17 2020 17:27:03 | Author: (none, default config)
Recv: Compiled: Feb 17 2020
Recv: echo: Free Memory: 1847  PlannerBufferBytes: 1792
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: FSensor ENABLED
Recv: echo:SD init fail
Recv: echo:SD init fail
Recv: ok
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Prusa-Firmware 3.9.0-RC1 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK2.5S EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: ok
Send: M21
Recv: echo:SD init fail
Recv: ok
Send: M117 192.168.1.204
Recv: ok
Send: M105
Recv: ok T:22.4 /0.0 B:22.5 /0.0 T0:22.4 /0.0 @:0 B@:0 P:132.3
Send: M105
Recv: ok T:22.4 /0.0 B:22.5 /0.0 T0:22.4 /0.0 @:0 B@:0 P:27.7
Send: M105
Recv: ok T:22.5 /0.0 B:22.5 /0.0 T0:22.5 /0.0 @:0 B@:0 P:22.6
Send: M105
Recv: ok T:22.5 /0.0 B:22.5 /0.0 T0:22.5 /0.0 @:0 B@:0 P:22.6
Send: M105
Recv: ok T:22.4 /0.0 B:22.5 /0.0 T0:22.4 /0.0 @:0 B@:0 P:22.6
Recv: MMU not responding - DISABLED
Send: M105
Recv: ok T:22.4 /0.0 B:22.5 /0.0 T0:22.4 /0.0 @:0 B@:0 P:22.6
Changing monitoring state from "Operational" to "Starting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Starting" to "Printing"
Send: N1 M73 P0 R11*36
Recv: NORMAL MODE: Percent done: 0; print time remaining in mins: 11
Recv: SILENT MODE: Percent done: 255; print time remaining in mins: -1
Recv: ok
Send: N2 M114*37
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0.00 Y:0.00 Z:0.00 E:0.00
Recv: ok
Send: N3 M201 X9000 Y9000 Z500 E10000*57
Recv: ok
Send: N4 M203 X500 Y500 Z12 E120*8
Recv: ok
Send: N5 M204 P2000 R1500 T2000*82
Recv: ok
Send: N6 M205 X10.00 Y10.00 Z0.20 E4.50*63
Recv: ok
Send: N7 M205 S0 T0*36
Recv: ok
Send: N8 M107*45
Recv: ok
Send: N9 M862.3 P "MK2.5S"*23
Recv: ok
Send: N10 M862.1 P0.6*89
Recv: ok
Send: N11 M115 U3.8.1*89
Recv: ok
Send: N12 G90*35
Recv: ok
Send: N13 M83*42
Recv: ok
Send: N14 M104 S215*86
Recv: ok
Send: N15 M140 S60*103
Recv: ok
Send: N16 M190 S60*105
Recv: T:22.50 E:0 B:22.7
Recv: T:22.50 E:0 B:22.7
Recv: T:22.56 E:0 B:22.7
Recv: T:23.00 E:0 B:22.8
Recv: T:23.47 E:0 B:23.0
Recv: T:24.31 E:0 B:23.0
Recv: T:25.34 E:0 B:23.0
Recv: T:27.09 E:0 B:23.1
Recv: T:29.09 E:0 B:23.1
Recv: T:31.12 E:0 B:23.1
Recv: T:33.62 E:0 B:23.2
Recv: T:36.10 E:0 B:23.4
Recv: T:38.88 E:0 B:23.7
Recv: T:42.03 E:0 B:23.7
Recv: T:44.55 E:0 B:23.7
Recv: T:47.95 E:0 B:23.9
Recv: T:50.74 E:0 B:23.9
Recv: T:53.21 E:0 B:23.7
Recv: T:56.51 E:0 B:24.0
Recv: T:59.28 E:0 B:24.2
Recv: T:61.82 E:0 B:24.5
Recv: T:64.95 E:0 B:24.6
Recv: T:67.52 E:0 B:24.7
Recv: T:70.78 E:0 B:24.9
Recv: T:73.20 E:0 B:25.1
Recv: T:75.69 E:0 B:25.3
Recv: T:78.85 E:0 B:25.4
Recv: T:81.32 E:0 B:25.6
Recv: T:83.71 E:0 B:25.9
Recv: T:86.66 E:0 B:25.9
Recv: T:89.05 E:0 B:26.1
Recv: T:91.38 E:0 B:26.2
Recv: T:94.27 E:0 B:26.3
Recv: T:96.51 E:0 B:26.7
Recv: T:99.34 E:0 B:26.8
Recv: T:101.63 E:0 B:27.0
Recv: T:103.89 E:0 B:27.2
Recv: T:106.72 E:0 B:27.4
Recv: T:108.96 E:0 B:27.6
Recv: T:111.18 E:0 B:27.8
Recv: T:113.91 E:0 B:28.1
Recv: T:116.06 E:0 B:28.2
Recv: T:118.14 E:0 B:28.5
Recv: T:120.75 E:0 B:28.6
Recv: T:122.95 E:0 B:28.9
Recv: T:125.57 E:0 B:29.1
Recv: T:127.61 E:0 B:29.5
Recv: T:129.61 E:0 B:29.5
Recv: T:132.24 E:0 B:29.9
Recv: Error:Temperature heated bed switched off. MINTEMP triggered !
Recv: // action:cancel
Changing monitoring state from "Printing" to "Error: Temperature heated bed switched off. MINTEMP triggered ! - // action:cancel"
Send: M112
Send: N17 M112*23
Send: N18 M104 T0 S0*24
Send: N19 M140 S0*93
Changing monitoring state from "Error: Temperature heated bed switched off. MINTEMP triggered ! - // action:cancel" to "Offline (Error: Temperature heated bed switched off. MINTEMP triggered ! - // action:cancel)"
Connection closed, closing down monitor
`

jjarts, 2020-04-08:
@DRracer Just chatted w/ Filip at Prusa Tech Support and he suggested I post here.  I'm on Firmware 3.9.0 RC1 and also getting the same mintemp bed error consistently on my mk2.5s MMU.  The error can be reproduced regularly.  It can also be avoided entirely by waiting a min or two in between initiating 'print from SD' and selecting the filament.  Maybe it just needs a little extra time to warm up during the safety check?

hernandez87v, 2020-10-07:
I just updated my MK2.5s to latest. (as a side note I've been running MK2.5 FIRMWARE because I havent fully upgraded my extruder but it has neve been an issue)

- FIRMWARE_NAME: prusa3d_fw_MK25_3_9_1_3518_RAMBo13a_en-es.zip
- previous FIRMWARE: prusa3d_fw_MK25_3_9_0-RC1_3272_RAMBo13a.zip

I was printing for almost 4hrs through Octoprint when I checked on the pinter is said:
- M112 called. Emergency Stop.

I put gcode on SD Card instead as suggested with Octo unplugged and the heatbed wont heat up. Light wont come on.  I notice this specific firmware says "Octoprint causing M112 fixed"

- wiggled the cable
- connected back to octo to check the log and this is what I get: 

`Recv: FIRMWARE_NAME:Prusa-Firmware 3.9.1 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK2.5 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: ok
Send: M20
Recv: Begin file list
Recv: KEYCHA~1.GCO 1744057
Recv: /TRASH-~1/FILES/4HRS33~1.GCO 7625337
Recv: /TRASH-~1/FILES/MK3_PL~1.GCO 786232
Recv: PLA_GE~1.GCO 25124635
Recv: PLA_PR~1.GCO 292205
Recv: PLA_TR~2.GCO 19192694
Recv: PLA_VA~1.GCO 7709051
Recv: V2CALI~1.GCO 785
Recv: PLA_3D~1.GCO 1817324
Recv: PLA_AD~1.GCO 20381063
Recv: PLA_BA~1.GCO 259965
Recv: PIM2_C~2.GCO 1325741
Recv: CALIBR~2.GCO 21286
Recv: GBA_LI~2.GCO 144102
Recv: 2HRS-S~1.GCO 5675108
Recv: 18MIN-~1.GCO 992508
Recv: 1HR41M~1.GCO 4954980
Recv: 1HR7MI~1.GCO 3137509
Recv: 6HRS40~1.GCO 14274353
Recv: FKS_03~1.GCO 2145839
Recv: NEW-CA~1.GCO 3052465
Recv: N_035M~1.GCO 810508
Recv: End file list
Recv: ok
Send: M117 IP 192.168.1.80
Recv: ok
Send: M117 192.168.1.80
Recv: ok
Send: M140 S50
Recv: ok
[...]
Recv: MMU not responding - DISABLED
[...]
Recv: Error:Heaters switched off. MINTEMP BED triggered!
Changing monitoring state from "Operational" to "Error: Heaters switched off. MINTEMP BED triggered!"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error: Heaters switched off. MINTEMP BED triggered!" to "Offline (Error: Heaters switched off. MINTEMP BED triggered!)"
Connection closed, closing down monitor`

Now im here asking myself why I upgraded lol

CWStevens, 2020-10-07:
Me too!  Mk3s

Prusa-Support, 2022-05-14:
Hello! Sorry for intervening on this thread so late. Unfortunately, I can only confirm the info from @DRracer: for MK2, MK2S, MK2.5, MK2.5S, it wouldn't be possible for the board to reliably read temps below 15 degrees C. 

@hernandez87v @CWStevens is the issue still actual? In your case it could have been related with faulty thermistor cable, in case of anything more to review please let me know via email info@prusa3d.com mentioning me in the subject ("attn: Alessandro"). I am closing the issue for now based on this info. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-14 by Prusa-Support

marcvs, 2026-01-07:
I do get this error in winter. The printer is located in my basement where it is colder in winter. I need to reset the winter after each startup, and pre-heat again. I didn't bother too much, but I think the problem is around for a couple years.
Making this check for faulty thermistor a bit less picky, should help fixing this issue.