prusa3d/Prusa-Firmware #2493: [FEATURE REQUEST] Save print for power-down

wpegden, 2020-02-14:
If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** MK3S (and MMU2S)

**Is your feature request related to a problem? Please describe.**
Although power-panic can be used to save a print state in a power-off scenario, there seems to be no way to save a print state to the internal memory other tricking the printer into thinking there is a power panic.
  
**Describe the solution you'd like**
There should be a menu item "save for shutdown" or similar, which:
(ideally:) moves the print head up off the item
saves the print state to internal memory
changes the LCD screen to inform the user to power off the printer.

Upon power on, the user is asked if they want to continue the previous print.

People are sometimes in the position of having to move a printer (and unplug it) during a long print, and currently the only approach is to trick the printer into thinking there is a power panic.

wavexx, 2020-02-26:
Once we fix #2325, it will be possible to just pause the print and unplug the printer. I think there's no cleaner way than that!

wpegden, 2020-02-26:
I agree that would be an improvement, but still, we are depending in this case on the power panic being detected in time.  The Prusa documentation, for example, warns that removing the plug may not always work:

wavexx, 2020-02-26:
The energy left in the PSU during a pause should always be sufficient (in the last RC release we also move the extruder out of the print just prior to shutting down).

The only usability concern I have is that on the MK3 you would need to pull the chord to trigger a panic - definitely not what I would expect, while the MK3S with the delta PSU works The Right Way™ just by flipping the switch.

So at least on the MK3, to avoid confusion, and explicit "save" in the pause menu could avoid issues..

nulaft, 2020-03-01:
Is there any downside to just always saving the print state when pausing a print?

wavexx, 2020-03-02:
eeprom wear, but maybe it's not such a concern ..

closed 2023-10-19 by github-actions[bot]