prusa3d/Prusa-Firmware #2495: [QUESTION] What is the routine for using the MMU2S Blade?

fromanza, 2020-02-14:
MK3S & MMU2S

1. What is the routine or intention on using the MMU2S blade?
2. If enabled through the settings menu will this cut the tips off during each filament change?
3. Will PrusaSlicer have options to use this function or is this explicitly through firmware for now?
4. Any update on the MMU2S part STL's update for dual blade or improved blade holding parts?

The "cut filament" in the menu seems to work quite well however I assume the bin will fill up quite quickly wih lots of changes.

leptun, 2020-02-14:
The filament is only cut automatically if it fails to load two times in a row.
There is also an experimental firmware option to always cut the filament, but it is only an experiment as of now.
The new blade holder uses only one blade. Only the blade position is moved slightly. The selector and blade holder will have to be reprinted.

fromanza, 2020-02-14:
Are these updated models available for download?

vintagepc, 2020-02-15:
I do not think so; there hasn't been any activity on the parts github for a while.

leptun, 2020-02-15:
Not yet. The parts are still being tested.

closed 2023-10-19 by github-actions[bot]