prusa3d/Prusa-Firmware #2496: [QUESTION] documentation or list of semicolon commands?

wpegden, 2020-02-14:
QUESTION:

I learned from github about three special commands which can be issued through the serial port:
;C32u2_RMD
(disable reset on serial connection)
;C32u2_RME
(enable reset on serial connection)
;C2560_RES
(immediately reset printer)

Are there other undocumented "semicolon" commands?  A google search for these turn up just a handful of discussion pages on prusa forums / issue trackers listing just these three examples.

[As far as I can tell, ;C2560_RES (or using the auto-rest upon serial connection) seems to be the only way to remotely progress with a printer which is "busy: paused for user".]

Thanks!!

vintagepc, 2020-02-15:
A quick google turns up this
https://cfl.prusa3d.com/display/PI3M3/UsbSerial+commands

Use at your own risk, etc, etc.

wpegden, 2020-02-15:
Thanks!  Not sure how I failed to find that.

closed 2020-02-15 by wpegden

easonjim, 2023-12-26:
@vintagepc hello vintagepc，Where can I find the code implementation? thanks.

vintagepc, 2023-12-26:
> @vintagepc hello vintagepc，Where can I find the code implementation? thanks.

AFAIK the bootloader code is not in a public repository.

easonjim, 2023-12-26:
@vintagepc thanks, I found this, is it this? https://github.com/prusa3d/rambo-32u2-usbserial/blob/master/Arduino-usbserial.c

vintagepc, 2023-12-26:
I think that's for the older Rambo (Mk2) boards given its age, so it may not be entirely accurate

easonjim, 2023-12-26:
@vintagepc  I get it, thank you very much.