prusa3d/Prusa-Firmware #2497: [BUG] Loading filament first phase speed in M600 and in command Load filament differs

ochm, 2020-02-15:
**Printer type** - MK3S

**Printer firmware version** - 3.9.0-beta-3175

**Describe the bug**
When the filament is loaded by click Load filament in menu while changing filament by M600 command, the speed of loading have two phases - quicker, and slower. The quicker is pretty good, 
and is IMHO till filament reach melting zone, the slower simulates extrusion. OK, no problem.

But when the Load filament is called from printer menu, the first phase of loading is slow. IMHO in firmware version  3.8.1 the first phase of loading was quicker than in version 3.9.0-beta-3175.

**To Reproduce**
Try to load filament friom menu Load Filament, and try to load filemant when M600 is issued.
  
**Expected behavior**
The same quick speed of Load filament in first phase when is Load filament called from menu (the same speed like is in M600 command).

leptun, 2020-02-15:
The slower speed is used for manual loading since the printer can’t assume if the filament is already fully loaded or not. If it would load faster and if there was filament in the hotend already when you press load, then horrible grinding would occur.
M600 doesn’t need this restriction since when the user inserts filament the printer is certain that there was no filament in the extruder, so it can load fast.

ochm, 2020-02-16:
I understand but i liked faster speed in manual l;oading in earlier versions. Sometimes i used it to check if the idler is tightened enough (the stepper lost stepn no grinding).
Another way for faster loading is to allow faster loading speed when the fsensor doesn't see filament loaded at the start of manual loading. So this looks like feature request now?

closed 2023-10-19 by github-actions[bot]