prusa3d/Prusa-Firmware #2500: [BUG] [3.9.0 RC1] Printer crashes/reboots when no filament is loaded in MMU

sgoll, 2020-02-18:
**Printer type** - MK3s
**Printer firmware version** - 3.9.0 RC1, 3.9.0 BETA

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**Describe the bug**

When starting the default Linear Advance calibration pattern from https://marlinfw.org/tools/lin_advance/k-factor.html, the printer resets just between 7x7 bed leveling and actually starting the print.

This only happens when no filament has been loaded to the nozzle (MMU2S). When I manually load some filament to the nozzle (via the menu), the print proceeds as normal.

The crash/reboot does not happen for the same file with firmware 3.8.1.
  
**To Reproduce**

- Generate Linear Advance calibration file from https://marlinfw.org/tools/lin_advance/k-factor.html (exact details do not matter, all defaults is sufficient to reproduce)
- Do _not_ feed filament to nozzle, just feed filament into MMU2S
- Start the print (with 7x7 bed leveling enabled)
  
**Expected behavior**

There should be no hard reset/reboot of the printer. If anything, the printer should display an error message.
  
**G-code**

```
; ### Marlin K-Factor Calibration Pattern ###
; -------------------------------------------
;
; Printer: mk3s
; Filament: pla
; Created: Sun Feb 16 2020 15:54:55 GMT+0100 (Central European Standard Time)
;
; Settings Printer:
; Filament Diameter = 1.75 mm
; Nozzle Diameter = 0.4 mm
; Nozzle Temperature = 205 °C
; Bed Temperature = 60 °C
; Retraction Distance = 1 mm
; Layer Height = 0.2 mm
; Z-axis Offset = 0 mm
;
; Settings Print Bed:
; Bed Shape = Rect
; Bed Size X = 200 mm
; Bed Size Y = 200 mm
; Origin Bed Center = false
;
; Settings Speed:
; Slow Printing Speed = 1200 mm/min
; Fast Printing Speed = 4200 mm/min
; Movement Speed = 7200 mm/min
; Retract Speed = 1800 mm/min
; Printing Acceleration = 500 mm/s^2
; Jerk X-axis =  firmware default
; Jerk Y-axis =  firmware default
; Jerk Z-axis =  firmware default
; Jerk Extruder =  firmware default
;
; Settings Pattern:
; Linear Advance Version = 1.5
; Starting Value Factor = 0
; Ending Value Factor = 2
; Factor Stepping = 0.2
; Test Line Spacing = 5 mm
; Test Line Length Slow = 20 mm
; Test Line Length Fast = 40 mm
; Print Pattern = Standard
; Print Frame = false
[... 247 lines cut ...]
G1 E-1 F1800 ; retract
G1 Z0.3 F1200 ; zHop
G1 X143 Y110.5 F7200 ; move to start
G1 Z0.2 F1200 ; zHop
G1 E1 F1800 ; un-retract
G1 X143 Y112.5 F7200 ; move to start
G1 X143 Y114.5 F7200 ; move to start
G1 X145 Y114.5 E0.0798 F1200 ; 2
G1 X145 Y112.5 E0.0798 F1200 ; 2
G1 X143 Y112.5 E0.0798 F1200 ; 2
G1 X143 Y110.5 E0.0798 F1200 ; 2
G1 X145 Y110.5 E0.0798 F1200 ; 2
G1 E-1 F1800 ; retract
G1 Z0.3 F1200 ; zHop
;
; finish
;
M104 S0 ; turn off hotend
M140 S0 ; turn off bed
G1 Z30 X200 Y200 F7200 ; move away from the print
M84 ; disable motors
M502 ; resets parameters from ROM
M501 ; resets parameters from EEPROM
;
```

leptun, 2020-02-18:
@sgoll The issue should be fixed by #2466 
@vintagepc is this correct?

vintagepc, 2020-02-18:
Yes, you are correct.

The underlying issue in that PR is that we hit an assert() during the runout routine if the current filament is undefined ("F?").  

There are a number of ways to get in that situation, and I believe this is another such path, since a non-MMU print does not invoke T-codes to set a tool, nor have the Tx/Tc pair to invoke manual selection, which are the normal ways the filament becomes set during a cold start.

sgoll, 2020-02-18:
@leptun @vintagepc Ah, sounds good. I'll close this then and test again when the next version gets released.

Thank you both!

PR #2466 (2020-03-26): #2464 fix runout for unknown filament (also fixes #1993, #2301)
This fixes the printer soft-resetting if a filament runout is detected while the current filament is unknown (e.g. user loaded to nozzle, and an event caused a power cycle (power panic during MMU print), or user turned off printer and continued printing with the same filament the next day). 

Because the current filament is unknown, we cannot reliably retract it and move on to the next one if spooljoin is enabled (theoretically we could try this and let the MMU try a retract/home/next filament, but that seems risky),  nor can we m [... cut]
closed 2020-02-18 by sgoll