prusa3d/Prusa-Firmware #2502: [BUG] Extra extrusion after filament change

espr14, 2020-02-18:
i3 MK3S FW 3.8.1

1. Filament runs out during print
2. Printer pauses and asks for filament change.
3. Filament is pulled out.
4. Filament is pulled in.
5. Answer for question is YES (everything OK).
6. Now it extrudes extra filament which is immediately attached to the printed object !!!

Should not extrude extra filament but do retraction. Since there's no default retraction length, there should be Marlin instruction added which defines default retraction length. There could be retraction instruction as well.

DRracer, 2020-02-19:
This extra purge is a feature and the user shall count with that. It is there to prime the extruder and to reduce (avoid) the gap in the print caused by missing filament. Removing excess filament from the print is less a problem than adding back missing filament.

espr14, 2020-02-19:
Missing filament is bad for external perimeter but not for the rest (in most cases). But this excess filament is attached to the print always. So, at least, it would be nice to turn fan to 100% while extruding this excess filament not to attach to the print well.

wavexx, 2020-02-26:
It was weird to understand this for me at first, but it works the following way: leave the excess attached to the nozzle while swapping the filament. As soon as the extruder starts moving, grab it with a quick motion to make a nice break at the nozzle. If you do that, the result is almost perfect (next to zero gap or blob).

You cannot "grab" the filament unless you have some excess being extruded.

Area5142, 2020-02-27:
I think this is related to the unfixed issue "M600 does unwanted extrude after 'is color clear?' confirmation" #197

I print almost only in PETG and this behaviour often makes the extra filament stick to the nozzle instead of the perimeter. The new lump on the nozzle is then dragged along making more harm than a tiny hole in the perimeter.

I have found no way to turn this "feature" off...

I understand that this extra purge is to build up the pressure in the nozzle - maybe there is another way to do that or reduce the problem:
1) Move nozzle to the spot where printing start before extruding (avoiding extrusion in the air).
2) No extra extrude, but let the new LA1.5 (Linear Advance) handle all pressure control in the nozzle.
3) Making a way to turn this feature off in the firmware.

Area5142, 2020-02-27:
Another relevant reference: "Blobs upon resume after M600" #904

wavexx, 2020-02-28:
The problem has already been described, it boils down to the fact that between "is color clear?" and the extruder moving some material has oozed already. We could retract just before the "is color clear?" prompt, but stuff like TPU would probably still leak a bit.

LA cannot do magic either. It has to assume the nozzle is fully primed in order to work.

Is it really such a big issue? When swapping filament some excess will always be there. I just pull it at the very end instead of doing it right after the prompt.

I also print a ton of PETG, so I know really well the effect you describe.

Area5142, 2020-02-28:
@wavexx 
I try to quickly grab the extra extruded filament, but when printing things like flat signs the room from the build plate to the nozzle is very small (only about 10mm). It would help a lot if the nozzle had a larger distance over the build plate - that would make room for multiple purges and cleaning the nozzle.
A short retraction (2mm) after "is color clear" would also help and properly prevent filament sticking to the nozzle.

espr14, 2020-03-03:
The amount of extra extrusion is very much. It's like 10 cm with 0.4 nozzle. This can lead to collision during printing. Even with 1 mm nozzle it is more than enough. Moreover, only 0.6 mm nozzle is supported.

GunSmoker, 2020-04-30:
This is definitely not a bug. I do remember instructions on changing filament, which explicitly says that you must grab extruded filament and hold it in place until hotend will move. Moving hotend will break filament from nozzle, leaving clean primed nozzle, and filament blob on your tweezers.

gogglespisano, 2020-05-18:
"I do remember instructions on changing filament, which explicitly says that you must grab extruded filament and hold it in place until hotend will move. Moving hotend will break filament from nozzle, leaving clean primed nozzle, and filament blob on your tweezers."

This doesn't work very well for me.  When I grab hold of the filament change blob if often breaks loose from the nozzle.

How does this help with priming and pressurizing?  The extra extrusion is into open air just like the filament change extrusion 2 seconds earlier.  The only need for this is if you wait a long time after the filament change to continue printing.  How about a timer? Less than a few seconds - no extrusion; Longer - do the extra extrusion.

sledjunk, 2020-08-12:
Sorry I posted #2801 as it is a double post.  However, I will add that after the filament change is done, and while the printer is moving to the start position, the extruder is turning, so it really is almost impossible to prevent a blob, especially with PETG.

wavexx, 2020-08-13:
Forgot about this until now.. I had a few filament swaps yesterday, and I might have to agree.

As @sledjunk said the extrusion keeps going until XY are in-line with the print. Sometimes grabbing the filament as late as possible is not always easy on large prints.

Thinking again, I'd still do an extrusion before starting to move (the filament just oozes out even if you're fast with the button), followed by a quick retract and then move. This way if you grab the last bit of filament before (or during) the extruder moves there's less ooze going on.

I tried this by simulating a change with some gcode and seems to work "good enough", certainly not worse than before even if you click "ok" after ~30s or so.

Some ooze is still present even with a retract though. The move back to the print is slowish.. I'm not sure if this speed is fully intentional?

sledjunk, 2020-08-13:
I tried again yesterday and I used a strip of plastic (actually a ruler that came with overture PETG) and held it to the nozzle when it landed on the job so that it caught most of the blob.  It helped somewhat, but you almost have to ruin one print so that you know where the nozzle is going to land on the job.

gogglespisano, 2020-08-13:
If there was a beep and 1 second pause just before moving back to print you could easily grab it.

This would be good at the start of prints too.  I've gotten used to grabbing the ooze just before a print starts by watching the temps and knowing when it's about to start.  If I don't grab the ooze, it's hit and miss if it will fall off during leveling and be laying in the middle of the print area.

sledjunk, 2020-08-13:
I haven't really had an issue with start of print because of the purge done after bed leveling and before the print begins

gogglespisano, 2020-08-13:
I don't have a problem after the purge starts, but once in a while the preheat ooze will fall off while leveling and then get into the print.  

It happens often enough that I got in the habit of grabbing in just before the leveling starts.

benr916, 2021-02-20:
Yea, this is still an issue even on the latest version firmware and PrusaSlicer. Ive gotten pretty good at grabbing that extra filament that leaks out but it would be nice if I didnt have to babysit the change over so much.

ChrisWhite1985, 2021-09-05:
I have this issue repeatedly on both my MK3S when using PLA.

I use an M600 Tool Change Command during the same layer to get multiple colours for signs and plaques. This extra purge is un-necessary as you have just wasted a load of filament flushing the old colour and making sure its ready. It just creates blobs sufficient that the nozzle crashes on the print and I've had it wreck production pieces over and over again.

Its not oozing, its not the wrong filament change technique (I do 20 changes on some prints and always grab the excess and pull it away at the very last second.). It just sits there and blows a blob of filament as if its trying to wipe.

Panayiotis-git, 2021-09-06:
I remember some time ago, when I did some prints with many color changes per layer (over 10), the only reliable way to do the filament change was to use at the slicer a fine tuned purge tower, even without the MMU.

ChrisWhite1985, 2021-09-06:
> I remember some time ago, when I did some prints with many color changes per layer (over 10), the only reliable way to do the filament change was to use at the slicer a fine tuned purge tower, even without the MMU.

Thanks, actually I always turn off the purge tower but now I can see the use of it for fixing this issue. Thanks Panay!

ChrisWhite1985, 2021-09-11:
> I remember some time ago, when I did some prints with many color changes per layer (over 10), the only reliable way to do the filament change was to use at the slicer a fine tuned purge tower, even without the MMU.

I tried this and it worked perfectly! No sign of any blobs! Thanks very much for the help.

floslo, 2021-09-20:
**Summary of the issue**

**The problem:** **unwanted blobs/strings of filament ruin prints for users** because there is very little time to remove the extra extruded string and it is **surprising and undocumented**.

**Surprising**: No extra extrusion into the air happens when loading filament at the start of a single color print.
**Undocumented**: Prusa "color change" help page [text](https://help.prusa3d.com/en/article/color-change_1687) / [video](https://youtu.be/F0IF5PAH7gs?t=26) does not mention / show the extra extrusion.

There are two suboptimal workarounds.
**Workaround 1**: Leave the first extrusion after filament change on the print bed, hold it with two fingers, press "yes" and when the extra extrusion happens remove all filament in one go. That's suboptimal because the extra extrusion string can break off and leave a little hanging from the nozzle and because the double extruded strings can twirl or bend back up and stick to the nozzle when printing PETG.

**Workaround 2**: Manually create a purge tower object in PrusaSlicer. Right click bed > "Add shape" > rectangle > size it about 3 cm long, 3 mm wide and height above the final color change layer. In the right pane list of objects drag the rectangle to the bottom to make it print first at each layer. Press "Slice now" and use the g-code preview sliders to double-check that print starts at the waste tower after color change. That's suboptimal because print time increases and the manual steps take time.

The problem has been reported multiple times for at least 4 years. 2017 [prusa forum](https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-software-archive/bug-color-change-nozzle-puke/) and issues years [2017](https://github.com/prusa3d/Original-Prusa-i3/issues/18), [2018](https://github.com/prusa3d/Prusa-Firmware/issues/904), [2020](https://github.com/prusa3d/Prusa-Firmware/issues/2502). New users still run into the problem - see comments on [Color change](https://help.prusa3d.com/en/article/color-change_1687) help page.

**Things that could fix or mitigate the problem**
1. Firmware: add 2 second delay after the extra extrusion to give the user more time to remove it before printing resumes.
2. Firmware: warn about the extra extrusion in LCD menu.
3. PrusaSlicer: In "color change" action add option to enable/disable the extra extrusion.
4. PrusaSlicer: In "color change" action warn about the extra extrusion.
5. Prusa "Color Change" [help page](https://help.prusa3d.com/en/article/color-change_1687). Inform about the extra extrusion and describe workaround 1 and 2.
6. Prusa "Color Change" [video](https://youtu.be/F0IF5PAH7gs?t=26). Update video to show the extra extrusion step and workarounds for it.

SquidPony, 2022-01-13:
Any chance this is going to ever get fixed?

If an extra purge extrusion is desired, users can add that in. We need a way to not have it push extra filament when it doesn't need to.

Like in comment https://github.com/prusa3d/Prusa-Firmware/issues/2502#issuecomment-913200978, I have a single extruder and do color changes and after the color change have to babysit the purge. Sometimes there's room to do Workaround 1 listed above, but not if I'm printing something that goes all the way to the resting edge where the hotend moves for the change.

camdenorrb, 2022-07-28:
I also experience this hm

Prusa-Support, 2022-08-08:
Hello! Thank you for the feedback on this. Unfortunately, no timeframe yet or confirmation of a possible fix at this moment. In a way that extra extrusion does ensure the correct pressure and flow are there. Did the workarounds proposed on previous comments improve the situation so far? Even just being ready with pliers to remove that excess can do, in addition to all the workarounds on the previous comments. 

Alessandro Pantaleo
Prusa Research

floslo, 2022-08-09:
@Prusa-Support What I [above](https://github.com/prusa3d/Prusa-Firmware/issues/2502#issuecomment-922870974) described as Workaround 1 helps a little, but a proper fix is still needed. Until a fix is in place Prusa could update the documentation text/video and PrusaSlicer to **inform and warn about the issue**? See my suggestions 4 5 6 at the linked comment.

Prusa-Support, 2022-09-01:
Hello! Thank you for the reply, noted this suggestion internally for us to further evaluate and take into consideration. In case of more info from our side we'll update on this thread, but in case of any more suggestions or aspects to review feel free to add more comments here. 

Alessandro Pantaleo
Prusa Research

ChrisWhite1985, 2022-09-02:
@Prusa-Support 

Please try this in your lab to re-create what we are getting and then you can work towards a solution?

Prepare a model in PrusaSlicer that has 2 colours on the same layer. I.e. a split model that you can assign 2 Extruders.

Go to Printer Settings > General  and select Extruders (param "extruders_count") to a value of 2.
Go to Printer Settings > Custom G-Code...

In the window for "Tool change G-Code" enter...
{if layer_z > 0}M600{endif}

Go to Print Settings > Multiple Extruders
Check Wipe Tower (param wipe_tower) is disabled / unticked.

In the Plater tab, assign the two parts on the same layer as Extruder 1 and Extruder 2 colours.

Run the print and watch for the M600 to trigger. The printer will print the first colour as normal. Then it will come forward for a filament change. After you have done the filament change, and held the excess purge from the nozzle as it moves across to the print - WATCH CAREFULLY. The nozzle moves down onto the print and blows a big bubble of filament, almost like a large de-retraction? Its happening in firmware so we can't see it in g-code? This does not happen when you implement a wipe tower.

puterboy, 2023-04-27:
> Foslo wrote
Things that could fix or mitigate the problem
1. Firmware: add 2 second delay after the extra extrusion to give the user more time to remove it before printing resumes.
2. Firmware: warn about the extra extrusion in LCD menu.
3. PrusaSlicer: In "color change" action add option to enable/disable the extra extrusion.
4. PrusaSlicer: In "color change" action warn about the extra extrusion.
5. Prusa "Color Change" [help page](https://help.prusa3d.com/en/article/color-change_1687). Inform about the extra extrusion and describe workaround 1 and 2.
6. Prusa "Color Change" [video](https://youtu.be/F0IF5PAH7gs?t=26). Update video to show the extra extrusion step and workarounds for it.

Here are a couple more solutions
8. (Similar to 1) move printhead higher during filament change to give more room (and time) to ensure extra extrusion is cleared)
9. (Extension of  3) Instead of making it a binary option to turn on/off extra extrusion, allow user to set the amount of extra extrusion. Thus 0= no extrusion. 5mm = current default. Other numbers would allow for more or less depending on the use case. Also, I would suggest this option occur under "Filament Settings" rather than having to change it each time under "color change", especially since behavior is likely to be filament-dependent
10. Remove from slicer and add to firmware so that extra extrusion only occurs if there is a more than x-second gap between extrusion of filament during filament change and when "yes" is pressed

Prusa-Support, 2023-05-29:
Hello, thank you for the extra suggestions. I apologize for the late response and for the long time passed, there are unfortunately at the moment other aspects with a higher priority, given the recently released products, but I passed the feedback and suggestions on to our devs, in case of more info from our side on these aspects, we'll certainly let you know on this thread. 

Alessandro Pantaleo
Prusa Research

MarcoRemy, 2023-06-05:
This issue really needs to be fixed! To swiftly remove the extra extrusion with tweezers after hitting "resume" while the head is already mooving back to print is a real hassle.

ChrisWhite1985, 2023-06-05:
> This issue really needs to be fixed! To swiftly remove the extra extrusion with tweezers after hitting "resume" while the head is already mooving back to print is a real hassle.

That's not the issue in discussion here.

The process of holding the filament end as the printer moves towards the part is a non-issue.

The problem is that when the toolhead returns to the print after M600, after it brings the nozzle down to where it left off, it pushes / purges a small amount of filament that causes a blob. However, if you use a wipe tower, the blob / purge does not occur, not even on the wipe tower.

iamstevedavis, 2023-06-05:
> > This issue really needs to be fixed! To swiftly remove the extra extrusion with tweezers after hitting "resume" while the head is already moving back to print is a real hassle.
> 
> That's not the issue in discussion here.
> 
> The process of holding the filament end as the printer moves towards the part is a non-issue and a standard practice on MK3S.
> 
> The problem is that when the toolhead returns to the print after M600, after it brings the nozzle down to where it left off, it pushes / purges a small amount of filament that causes a blob. However, if you use a wipe tower, the blob / purge does not occur, not even on the wipe tower.

100%, good description.

Prusa-Support, 2023-06-27:
Thank you again for your input.

The current behavior may cause small artifacts like extra filament outside the object but no gaps, in fact, there is a chance that the user will let the nozzle ooze for some time before resuming and there would be no work around that.
You may want to delay the print resume intentionally or remove excess filament outside the object when the print is over.

Our developers will do their considerations about this.
For now, letting the nozzle ooze for several seconds before picking the purged plastic (and resuming) is the recommended workaround  - recommended, not necessary.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-06-28:
>The problem is that when the toolhead returns to the print after M600, after it brings the nozzle down to where it left off

On a side note, to my understanding, this specific aspect is a known bug on the Prusa Slicer side (https://github.com/prusa3d/PrusaSlicer/issues/4432) and they are currently scripting an adjustment.

Michele Moramarco
Prusa Research

ChrisWhite1985, 2023-10-11:
Can't be closed as I have not seen a reported fix. Not sure if this plagues the MK4 as well?

floslo, 2023-10-11:
@ChrisWhite1985 
> The process of holding the filament end as the printer moves towards the part is a non-issue and a standard practice on MK3S.
>
> The problem is that when the toolhead returns to the print after M600, [...]

There may be (at least) two issues here.

Holding the filament end after filament change is extra tricky in this situation because of the *unexpected and undocumented* extra extrusion right before the move. That is an issue in that plenty of people have reported it surprised them and ruined their prints. The practice of holding the filament is not documented by Prusa so I would not call it standard practice either.

As I wrote upthread https://github.com/prusa3d/Prusa-Firmware/issues/2502#issuecomment-922870974
until a fix is in place Prusa could update the documentation text/video and PrusaSlicer to inform and warn about the issue. See my suggestions 4 5 6 at the linked comment.

The help page video is misleading because it does not show the step where the extra extrusion happens. It gives the impression that you only have to press the dial to accept and the printer will do the rest. It does not show the practice of holding the filament down. 6 years after the first reporting of this issue Prusa still haven't taken even the quick and simple mitigation step of updating the help page and video.

Prusa-Support, 2023-11-22:
Thanks for showing care for the issue and keeping it "alive".

Our developers will consider improving the filament change instructions, and we don't exclude little operation adjustments to improve the current behavior - which is not buggy but necessary.

Michele Moramarco
Prusa Research

floslo, 2023-11-22:
> the current behavior ... which is not buggy but necessary.

Please explain how the current behaviour, specifically the unexpected extrusion of extra filament _after_ the user has loaded the new color filament and pressed "yes" on the "changed correctly?" question.

Until a clear such explanation has been given this issue should keep the bug label.

Because that behaviour differs from when the user loads filament at the beginning of single color prints: the menu asks, roughly, if the filament loaded correctly and the user can select "yes" to start the print (not extra unexpected filament is extruded here!) or "no" to extrude a bit more filament.

Prusa-Support, 2023-12-13:
The reason is explained in the first comment after the main post, and further again at https://github.com/prusa3d/Prusa-Firmware/issues/2502#issuecomment-1610126974.
Our developers will think about improving the feature.

Michele Moramarco
Prusa Research

Area5142, 2023-12-14:
I have been following this issue and I agree with the original poster that the extra extrusion after filament change is problematic. It often creates a blob on the surface of the print, which requires post-processing to remove. I understand that the extra extrusion is meant to prevent under-extrusion at the model return point, but I think that is a less noticeable issue than the blob.

I would prefer to have an option to disable the extra extrusion, or at least to adjust the amount of it. I think this would give more control to the user and allow them to fine-tune the filament change process according to their preferences and needs.

I appreciate the work that the developers are doing to improve the firmware and I hope that this issue can be resolved soon. Thank you.

MarcoRemy, 2024-02-12:
This is already a great improvement.
https://github.com/prusa3d/PrusaSlicer/pull/9036

Prusa-Support, 2024-02-17:
A _partially related issue_ where the nozzle moves back to the last position on the print - and releases some different color oozed filament in the wrong location - will be addressed via Prusa Slicer (issue https://github.com/prusa3d/PrusaSlicer/issues/2672, PR https://github.com/prusa3d/PrusaSlicer/pull/9036, commit https://github.com/prusa3d/PrusaSlicer/commit/540b800a5c1c76b25a809aac18f319e6a23093b7) .

The specific issue discussed in this thread instead - about  _the extra extrusion after filament change_  - should be solved with PR https://github.com/prusa3d/Prusa-Firmware/pull/4534. However please notice, even if merged, this feature is not enabled by default because experimental. Please go through the Pull Requst to learn more and the feature should be available in the next firmware release (but not active by default).

Michele Moramarco
Prusa Research

3d-gussner, 2024-03-13:
Closing as no interaction

floslo, 2024-03-13:
Please reopen this issue, it is **not** completed. The issue has been reported since 2017, **so around 7 years now**. Do you really need us to post a new comment saying the issue still exists every 30 days?

floslo, 2024-03-13:
> The reason is explained in the first comment after the main post, and further again at [#2502 (comment)](https://github.com/prusa3d/Prusa-Firmware/issues/2502#issuecomment-1610126974). Our developers will think about improving the feature.

@Prusa-Support no it is not explained in those comments.

The thing that is unexplained is why this extra (surprising and unwanted) extrusion of filament happens when manually changing filament color during a print *after* then new filament is loaded and some of it has already been extruded and directly *after* the user presses the button to resume the print with the new color.

After that button press it takes around 1 second for the printer to move from the new filament loading position near the front right corner to whatever location the to-be-resumed print is at. Why is it according to you necessary to extrude extra filament in the air during that move? If your concern is that after loading the new filament the user waited too long before pressing the button to resume and that causes some build up of heated filament that has to be disposed to avoid a blob on the real print then a proper fix would be to keep track of the time waiting for the user button press and, if too long, make the button press trigger some extrusion without moving and show a new button press prompt.

Note also the difference between this extra extrusion and what happens when starting a regular print: on a regular print start, after extruding a line of filament near the front left edge of the plate the head moves to the start print location and only then extrudes filament. It does *not* extrude any extra filament in the air between the front left edge and the start print location. For some reason *that* is not necessary - but a very similar extra air extrusion is claimed to be necessary after a filament color change?

Finally let me ask Michele Moramarco and other prusa support people who comment in this thread, have you yourself actually reproduced the issue on a printer and seen it with your own eyes?

puterboy, 2024-03-13:
> @Prusa-Support no it is not explained in those comments.
>
> The thing that is unexplained is why this extra (surprising and unwanted) extrusion of filament happens when manually changing filament [...]

I STRONGLY ENDORSE THIS COMMENT.
This is clearly BROKEN and the BIGGEST PITA when it comes to filament changes. In fact, it causes me to think twice about any designs with filament changes. In my several years of experience with the MK3S+, this "feature" (i.e. bug) only causes problems and is totally needless.

Why do the devs REFUSE to fix this despite numerous complaints and multiple proposed fixes ranging from simple to complex?

PLEASE FIX THIS AND PLEASE STOP TRYING TO CLOSE THIS BUG EVERY 30 DAYS!!!!!!!!!!!

puterboy, 2024-03-13:
BTW, the claim that the extra extrusion is "needed" for filament changes in case the user stepped away from the printer and the filament needs to be pressurized is BOGUS or at least is inconsistent with how Pauses are handled. So, I call BS.

Specifically, when a Pause is triggered followed by a Resume, there is no similar "extra extrusion" of filament -- even though it is very common that a person may fail to immediately respond to a Pause, potentially causing real depressurization of the extruder.
In contrast, it is highly unlikely to the point of ABSURDITY that a person will change the filament and then immediately forget about it and walkaway for a long period before returning to push the button.
This is back-asswards!!!

3d-gussner, 2024-03-14:
@floslo @puterboy Thanks for your feedback.

Did you read https://github.com/prusa3d/Prusa-Firmware/issues/2502#issuecomment-1950241112 ?
A community member addressed this and we even merged it so users having an issue with this can compile the firmware.

Also please check my comment in the PR https://github.com/prusa3d/Prusa-Firmware/pull/4534#issuecomment-1857721419

You may think of that extra extrusion as a bug and annoying but most are okay with it and "know" to grab the extra extrusion before it moves back to the print to ensure the print continues without any gaps.
If the "ooze" prevention doesn't work correctly you will see under-extrusion and maybe even gaps in the print, which is not acceptable.

So feel free to activate `COMMUNITY_PREVENT_OOZE` and compile the firmware.

You need to understand that changes like this have a big impact as the ooze prevention may work for a setup of material, nozzle size, etc. but needs different settings for other combinations.

We will not change it for the time being in the stock firmware.

3d-gussner, 2024-03-14:
@puterboy Please stay polite

>This is back-asswards!!!

That kind of comments do not help at all and I tend to step away from these kind of issue tickets.

ChrisWhite1985, 2024-03-14:
@3d-gussner - You have got it wrong. This is not about before the move back to the print, it's AFTER the move.

The nozzle comes down onto the print to continue printing and blows a bubble of filament unnecessarily before it continues printing creating a crash risk / print artifact. It does not do this when a purge tower is deployed.

Based on your response, you have never tested this to see what we are referring.

3d-gussner, 2024-03-14:
@ChrisWhite1985 I never had an issue during manual color changes and never saw the "blows a bubble of filament" on my prints.
I give it another try and look closely.

ChrisWhite1985, 2024-03-14:
@3d-gussner Make sure this is during a single layer colour change. I.e. 2 colours on same layer with manual filament change M600.

3d-gussner, 2024-03-14:
Here my test scenario:
- simple one layer 25x25x0.2 square gcode generated by PrusaSlicer 2.7.0+linux-x64-GTK3
- MK3 branch firmware (development towards FW3.14.0)
- SIM MK404 with high res extrusion to see extrusion width 
  - Star print
  - Send `M600`
  - Note the MK404 E step
  - Use `cpulimit -e M404 -l 5` to really slow it down and take notes and pictures

1. At "Please press the knob to unload ..." M404 E = `32.2642`
2. At "Was filament unload successful?" MK404 E = `-47.724`
3. At "Changed correctly" MK404 E = `47.9500` X = `211.000` Y = `0.0000` Z =`28.1124` (without MBL `27.20`)
4. While moving to X = `136.380` Y = `95.6699`  MK404 E = `52.2750` with Z = `28.0825` (without MBL `27.20`)
5. No MK404 E changes until Z =  `0.9500` (without MBL `0.20`)
6. The MK404 E only moves together with the X and Y asixs movements.

My conclusion:
- After confirming that the filament has been correctly changed the extruder extrudes about 5mm at the change height moving towards the xy position.
- During the descend from Z `27.20` to `0.2` **NO extra filament is extruded**
- Filament is only extruded at the 0.2 layer together with xy movements
 
I will try to prepare a animated gif.

[image]

3d-gussner, 2024-03-14:
As the MK404 SIM doesn't simulate nozzle pressure I will try that on the real hardware later.
Keep you posted.

floslo, 2024-03-14:
@3d-gussner thanks for reopening and testing!

> You may think of that extra extrusion as a bug and annoying but most are okay with it and "know" to grab the extra extrusion before it moves back to the print to ensure the print continues without [...]

How do you know most are ok with it?

How do you know most know the practical steps to deal with it?

As I noted upthread https://github.com/prusa3d/Prusa-Firmware/issues/2502#issuecomment-922870974 the issue is surprising because it is not documented. So how would someone who for the first time attemps a mid print filament color change know that the issue will happen or how to deal with it?

Let me repeat two earlier suggestions on what Prusa can do to mitigate

- Update Prusa "Color Change" [help page](https://help.prusa3d.com/en/article/color-change_1687) to inform that the extra ~5mm extrusion will happens and describe how you think the user should deal with it.

- Update Prusa "Color Change" [video](https://youtu.be/F0IF5PAH7gs?t=26) to show the ~5mm extrusion and how to deal with it. The current video is a quick feature teaser video rather than step by step help. It shows the "changed correctly?" button press at 0:33 but then immediately cuts to the finished print job. So it does not help the user with this issue - on the contrary it can give the user the expectation that there is no further user step to perform after pressing that button, making the issue even more surprising.

The most popular pratical method to handle the issue from this thread can be summarized as:

1 Let the first, automatically extruded larger amount of filament hang between the nozzle and the bed
2 put a finger on that filament on the bed to hold it in place
3 press the button to make the extra 5mm filament extrude onto the same string of filament you're holding
 
to which I add

4 .... and *hope* that as the head moves that extra 5mm will break off clean from the nozzle and stick to the filament string you're holding down. If it doesn't then react with lightning speed and move your hands to grab a piece of paper and wipe the dangling filament residue from the heated nozzle tip during the ~1s it is moving from the front right back to the print object. Make sure you don't burn your hand on the hot nozzle during this very quick hand movement. Also make sure you don't accidentally bump into the print job. If you fail your print may look ugly or be ruined.

Because in my experience sometimes that ~5mm _will_ stick to the nozzle. Especially when printing PETG that extra filament can twirl or bend back up and stick.

Lastly apologies if I'm missing something but can you explain why that extra ~5mm is necessary in this situation but not at the start of a regular print (right after the printer has placed a line of filament at the front left edge of the plate and while it is moving to the print start location)? I'm asking because the time it takes for the head to move is roughly the same in both those situations. If that difference can be explained in a short but clear way then consider also adding that explanation to the help page, so users can learn.

3d-gussner, 2024-03-14:
@floslo Thanks for the suggestion, I will forward it to the content team.

>right after the printer has placed a line of filament at the front left edge of the plate and while it is moving to the print start location

I am not the Prusa Slicer settings Pro but here my thoughts:
There is a slight difference between these two
- extruding filament on the plate builds up some pressure in the nozzle as the sheet "pushes" back
  - same applies to the purge tower 
- extruding filament mid air doesn't build up the same pressure.

floslo, 2024-03-14:
@3d-gussner  thanks

Pliers vs finger: For me using a finger (in glove if the user feels the plate heat is uncomfortable) is less error prone. With pliers I've accidentally moved the filament as I grabbed it, causing it to break off or weaken at the nozzle before the button press, which in turn caused the extra 5mm to not stick to the rest of the grabbed filament. That the success of this user step is sensitive to very small hand movements underscores that the best solution would be to find a way to get rid of this step.

> There is a slight difference between these two
>     * extruding filament on the plate builds up some pressure in the nozzle as the sheet "pushes" back
>       * same applies to the purge tower
>     * extruding filament mid air doesn't build up the same pressure.

Ok, that does make sense. If that is indeed the difference then it also suggests another possible solution to consider: in jobs where the user only changes color once and where there are no print objects near the front right part of the plate then instead of the extra 5mm extrusion in air make the printer extrude a line on the plate just like when starting a regular print, only now along the front right side of the plate. PrusaSlicer could be made to detect if a job is like that. My guess that would cover a majority of color change jobs. For example a user printing some object near the center of the plate and wants embossed text on top in a contrast color.

jypma, 2024-03-14:
Alternatively, if the current layer has enough infilled area available, could the extra 5mm extrusion perhaps be made into the infill area? There'd be some structure present to conveniently "wipe" the nozzle against, and it would be invisible since it's infill.

3d-gussner, 2024-03-14:
> Alternatively, if the current layer has enough infilled area available, could the extra 5mm extrusion perhaps be made into the infill area? There'd be some structure present to conveniently "wipe" the nozzle against, and it would be invisible since it's infill.

I had the same idea to define per sliced layer a safe ooze location in the empty space of the infill, but I got few comments on that:
- Needs development on Slicer
- Doesn't work on 100% infill layers (like bottom and top layers)
  - Color changes are often used on bottom layers to have some multicolor surface
- wiping in the infill may cause some blobs that may cause crashes.

[3 more replies]