prusa3d/Prusa-Firmware #2503: [BUG] mk3s firmware 3.8.1 printer does not pause when filament runs out

GSE0, 2020-02-19:
IR sensor turns from 1 to 0 when filament runs out but the printer continues. Sometimes pressing the knob will trigger a filament run out pause after the filament has actually run out, sometimes not. Factory reset and firmware flash did not change the behavior.

leptun, 2020-02-19:
Filament runout detection is disabled in the Sensors menu. This is so that people can see if the sensor works without triggering the firmware functions. In any other menu the function works.

GSE0, 2020-02-19:
Leptun: I have Fil. sensor [on] in the settings menu and this problem persists. Do you mean the sensor info menu under support? That is where I observe the IR sensor status changing from 1 to 0, but I do not see an option to turn off or on filament runout detection there. Please describe which menu you are referring to specifically. Thank you.

leptun, 2020-02-19:
When you are watching the filament sensor state from the “Sensor Info” menu, the runout and autoload functions are disabled regardless of “Fil. sensor” setting.
If you want to test for runout (and use it), stay on the status screen (or any other menu).
The “runout detection” option IS “Fil. sensor”.

GSE0, 2020-02-19:
OK - thanks that is good to know. I will try again in the office tomorrow. However, I only started looking at the sensor info screen because the filament runout detection failed when printing overnight and running out of filament twice. In both cases the printer LCD was on the default info screen.

GSE0, 2020-02-19:
OK it looks like the pause is working as intended now when filament runs out. There may have been some problem that was fixed by the factory reset and firmware flash which seemed to still be happening because I was watching the IR sensor status during debugging. Either that, or the issue is intermittent. Either way, I cannot easily reproduce it at this time. Thanks.

closed 2020-02-19 by GSE0