prusa3d/Prusa-Firmware #2504: [ENHANCEMENT]Remove "Remove filament" sign automatically

espr14, 2020-02-20:
Prusa i3 MK3S FW 3.8.1

1. Have filament in the extruder
2. Click on "Unload filament"
3. "Remove filament" sign appears and stays until user clicks

If the sensor of filament is 1 when the sign appears, FW could check it periodically and once it turns to 0, it could remove the sign since the filament is removed by the user. Thus, user does not need to click to remove the sign manually.

leptun, 2020-02-20:
I’m pretty sure it goes away automatically after a short period of time. I’ll check later to confirm this is the case.

jvpernis, 2020-04-17:
Just tried this with my MK3S MMU2S when it ran out of filament and I left it for over 2 minutes to beep after I had removed the filament, but nothing happened. Would indeed be nice if it were to detect when the filament has been removed and automatically continue to load the new filament.

closed 2023-10-19 by github-actions[bot]