prusa3d/Prusa-Firmware #2505: [ENHANCEMENT]Improve recovery after collision detection

espr14, 2020-02-20:
Prusa i3 MK3S FW 3.8.1

Collision detection can be triggered by:
- collision of print head (nozzle, PINDA, fan shroud) and printed model
- collision of balls in bearings
- stuck filament

Collision detection could be correct but sometimes it recovers badly so it continues printing somewhere else. Therefore, during recovery, printer should move X and Y axes to the start and to the end. If the length of axes are correct then it can continue printing. Otherwise, it should stop and show text like:
- "Print head cannot move freely. Loose filament or lubricate X axis."
- "Bed cannot move freely. Check for obstacles or lubricate Y axis ."

These tests could be part of self-test and an independent test to avoid future issues.

leptun, 2020-02-20:
This test is already part of selftest

HansPeterHaastrup, 2020-04-14:
This really is a two-part question - the latter was answered.
The first part 'Collision detection could be correct but sometimes it recovers badly so it continues printing somewhere else' is still very much a problem here

I'm interested in the funtion:
`void restore_print_from_ram_and_continue(float e_move)`

By the looks of it there doesn't seem to be any correction for any positional mistake here?
the problem (in my case) stems from a stuck spool of filament thus meaning that the position was not saved correctly.

it seems that the code assumes that it can move freely - it only checks if the fans can move the way they should.

The position is reloaded with the following:
`//first move print head in XY to the saved position:
	plan_buffer_line(saved_pos[X_AXIS], saved_pos[Y_AXIS], current_position[Z_AXIS], saved_pos[E_AXIS] - e_move, homing_feedrate[Z_AXIS]/13, active_extruder);
	//then move Z
	plan_buffer_line(saved_pos[X_AXIS], saved_pos[Y_AXIS], saved_pos[Z_AXIS], saved_pos[E_AXIS] - e_move, homing_feedrate[Z_AXIS]/13, active_extruder);
	//and finaly unretract (35mm/s)
	plan_buffer_line(saved_pos[X_AXIS], saved_pos[Y_AXIS], saved_pos[Z_AXIS], saved_pos[E_AXIS], FILAMENTCHANGE_RFEED, active_extruder);
	st_synchronize();`

I would argue that we need to do a homing on X and Y axis BEFORE we can continue the print.

espr14, 2020-04-14:
The collision is detected on a single axis so homing of the axis is sufficient. However, do we know that the homing was OK or there was another collision during homing? The latter collision causes bad re-calibration which causes shifted print.

Probability that the second collision occurs 1 milimeter or less from the start of the axis is very very low. So measuring the length of the axis (with 1 milimeter hysteresis) is sufficient to gain high probability that collision recovery was correct.

In other words, the collision was detected at e.g. X =143 mm. Then there's a recalibration which tells that the new start of the X axis is shifted by e.g. 28 mm and the print continues happily. This looks weird since the motor would have to loose so many steps. The collision detector would stop it much sooner.

HansPeterHaastrup, 2020-04-14:
The problem is that the collision detector doesn't stop it.
It seems that the restoring move is not governed by the collision detector:
I recover from a failed overnight print (the printer have 'crash detected, resume print?' in the display.
This crash came from the filament spool locking and hindering the free movement of the X axis. part itself was just fine.

I select 'Yes' and the printer resumes. However, the X-axis was not in home position and thus the x axis crashes into the far side of the printer. motor stalls and makes a horrible noise - no crash detected here. then the printing continues and of course the position is off.

I'm pretty sure that this can be replicated fairly easy. I should make a movie of this.

Prusa-Support, 2022-06-30:
Thank you for sharing your ideas.

In rare cases, an axis length check after a crash detection could actually help verify the axes are clear and smooth.

An already existing safety feature allows the print to pause if a few subsequent crashes are detected. For this to work, the printer needs to crash continuously. The existing feature would interrupt a print that, without user intervention, is doomed to fail (e.g. limited movement due to a tangled filament, or an external object fallen on the axis...).
The safety pause though may not be triggered if the obstacle that causes crashes is near the opposite end of the axis (no crashes as long as the printer doesn't reach the end of the axis), or until the nozzle trips over a print due to an "over/under-stepping axis" (e.g. debris between motor and pulley... / loosen motor pulley, deformed belt...) or due to a lifted portion of the print...

An axis length check after the crash could actually be an additional safety layer in some cases.

In my opinion, this sounds like a legit enhancement for the crash detection.
I'd leave this issue open as input for possible future improvements of the firmware.

Michele Moramarco
Prusa Research

espr14, 2022-06-30:
Meanwhile, I've already implemented this in the 32 bit (MINI). And since the 32 bit FW will be used in new printers (XL, AFS etc.) then it does not matter much for MK3. You can leave it here or close it.

closed 2023-10-19 by github-actions[bot]