prusa3d/Prusa-Firmware #2507: [BUG] Random power panic reboots in 3.9.0-RC1

vaxxi, 2020-02-21:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3
**Printer firmware version**-  3.9.0-RC1-3272

**Describe the bug**
Printer randomly reboots with what seems to be a power panic restart. This did not occur in 3.9.0-beta-3175.
  
**To Reproduce**
Start a print (ABS, so probably a bit more strain on the PSU); during the first layer, the printer beeps and reboots in random spots during the first layer (happened twice in a row).

Octoprint logs:

```
Send: N3125 M105*18
Recv: ok T:255.0 /255.0 B:110.1 /110.0 T0:255.0 /255.0 @:36 B@:60 P:73.6 A:44.7
Send: N3126 G1 X188.979 Y203.775 E0.01586*92
Recv: ok
Send: N3127 G1 X6.225 Y21.021 E7.89190*104
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: INT4
Recv: UVLO - end 426
Recv: echo:busy: processing
Recv: start
Printer sent 'start' while printing. External reset? Aborting job since printer lost state.
Send: N0 M110 N0*125
Changing monitoring state from "Printing" to "Cancelling"
Recv: echo: 3.9.0-RC1-3272
Recv: echo: Last Updated: Feb 17 2020 17:27:33 | Author: (none, default config)
Recv: Compiled: Feb 17 2020
Recv: echo: Free Memory: 1632  PlannerBufferBytes: 1792
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: CrashDetect DISABLED
Recv: PAT9125_RES_X=0
Recv: PAT9125_RES_Y=240
Recv: PAT9125_init:1
Recv: FSensor DISABLED
Recv: echo:SD init fail
Recv: Error:No Line Number with checksum, Last Line: 0
Recv: Resend: 1
Send: N0 M110 N0*125
```

```
Send: N2012 M105*22
Recv: ok T:255.6 /255.0 B:110.0 /110.0 T0:255.6 /255.0 @:25 B@:75 P:73.0 A:45.0
Send: N2013 G1 X126.978 Y6.225 E0.01586*88
Recv: ok
Send: N2014 M105*16
Recv: ok T:254.4 /255.0 B:110.0 /110.0 T0:254.4 /255.0 @:45 B@:81 P:73.0 A:45.0
Send: N2015 G1 X243.775 Y123.022 E5.04368*88
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: INT4
Recv: UVLO - end 355
Recv: start
Printer sent 'start' while printing. External reset? Aborting job since printer lost state.
Send: N0 M110 N0*125
Changing monitoring state from "Printing" to "Cancelling"
Recv: echo: 3.9.0-RC1-3272
Recv: echo: Last Updated: Feb 17 2020 17:27:33 | Author: (none, default config)
Recv: Compiled: Feb 17 2020
Recv: echo: Free Memory: 1632  PlannerBufferBytes: 1792
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: CrashDetect DISABLED
Recv: PAT9125_RES_X=0
Recv: PAT9125_RES_Y=240
Recv: PAT9125_init:1
Recv: FSensor DISABLED
Recv: echo:SD init fail
Recv: Error:No Line Number with checksum, Last Line: 0
Send: N0 M110 N0*125
Recv: Resend: 1
Recv: ok
```
**Expected behavior**
Don't reboot.
  
**G-code**
Doesn't seem to matter.

leptun, 2020-02-21:
Do you happen to use an UPS with the printer?

vaxxi, 2020-02-21:
No, connected directly to a 220V socket.

leptun, 2020-02-21:
It sounds a lot like a hardware issue to me. The MMU changes in the latest releases will not create this kind of issue.

vaxxi, 2020-02-21:
I would normally agree with you (especially since the printer's getting quite old now). However, the exact same GCODE (a ~5 hour print) printed just fine using the previous 3.9.0-beta firmware (using the same filament/build plate/power socket/room temperature).

I do see 3 commits related to power panic [between 3.9.0-RC1 and 3.9.0-beta](https://github.com/prusa3d/Prusa-Firmware/compare/v3.9.0-beta...v3.9.0-RC1):

[7f3d4a8491acabbbb139a41e4c5bff7e4dad9d3f](https://github.com/prusa3d/Prusa-Firmware/commit/7f3d4a8491acabbbb139a41e4c5bff7e4dad9d3f)
[5d8857371132b311c76d07c19be548a73281b8f0](https://github.com/prusa3d/Prusa-Firmware/commit/5d8857371132b311c76d07c19be548a73281b8f0)
[02d1525445b81002cc591738d69cbdb05988ee18](https://github.com/prusa3d/Prusa-Firmware/commit/02d1525445b81002cc591738d69cbdb05988ee18)

but not that experienced with the code to tell which one might cause this. Will test further.

leptun, 2020-02-21:
Try to switch back to the version that worked and confirm if it is still an issue.

vaxxi, 2020-02-21:
That's what I said above :) same print in the same conditions worked just fine with `prusa3d_fw_MK3_3_9_0-beta_3175` firmware. I'll upgrade to RC1 again and retest (just as an information tidbit, when the printer rebooted, the message in commit `02d1525445b81002cc591738d69cbdb05988ee18` did not appear).

leptun, 2020-02-21:
@vaxxi Ah. ok. I thought that you only tested the previous version before the issue started.
The message is only if the printer somehow manages to reboot while there is still some leftover power in the PSU.

@wavexx Any idea what might cause this?

wavexx, 2020-02-21:
The "POWER PANIC DETECTED" is only displayed for the "uvlo tiny" (second PP in a row). This is a full power panic.

I somehow doubt the interrupt is triggering randomly, I've been using RC1 for a while now (but .. surprises are always there).

I'd check/wiggle the cable connector on the rambo and a second step also under the psu to see if you can trigger a powerpanic just by doing so. Considering the vibrations that would be my first suspect.

If the panic still occurs on RC1, try disconnecting the powerpanic cable. If the panics go away, it's a hardware issue. If you still get panics, it's likely software.

ChrisVahi, 2020-04-07:
I am also experiencing the exact same condition. Random reboot after "start" sent from printer during a print. Seems to happen once the extruder completes final move to 0,0 of the mesh bed leveling measurement sequence. Here's the critical part of the serial log, complete print log attached as well.

FW 3.9.0-RC1 compiled using PF-build script

[3_9_0-RC1_bug_report_serial_log.txt](https://github.com/prusa3d/Prusa-Firmware/files/4442164/3_9_0-RC1_bug_report_serial_log.txt)

pekote, 2020-07-24:
Same problem here

Printer type - MK3s
Printer firmware version- 3.9.0-RC1-3421

Send: N8638 G1 X119.525 Y97.113 E0.02722*100
Recv: ok
Send: N8639 G1 X120.026 Y96.788 E0.03391*97
Recv: ok
Send: N8640 G1 X120.547 Y96.494 E0.03391*99
Recv: ok
Send: N8641 G1 X121.084 Y96.234 E0.03391*101
Recv: ok
Recv: INT4
Send: N8642 G1 X121.636 Y96.008 E0.03391*100
Recv: UVLO - end 564
Recv: start
Printer sent 'start' while printing. External reset? Aborting job since printer lost state.
Send: N0 M110 N0*125
Changing monitoring state from "Printing" to "Cancelling"
Recv: echo: 3.9.0-3421
Recv: echo: Last Updated: May 18 2020 17:11:15 | Author: (none, default config)
Recv: Compiled: May 18 2020
Recv: echo: Free Memory: 1675  PlannerBufferBytes: 1792
Recv: echo:Hardcoded Default Settings Loaded
Recv: adc_init
Recv: CrashDetect ENABLED!
Recv: FSensor DISABLED (sensor board revision: 0.3 or older)
Recv: Sending 0xFF
Recv: echo:SD card ok
Recv: Error:No Line Number with checksum, Last Line: 0
Send: N0 M110 N0*125
Recv: Resend: 1
Recv: ok
Send: N1 M113 S2*96
Recv: ok
Send: N2 M108*40
Recv: ok

Printer halt and reset and shows the above messages from OCTOPRINT. 

If I print from SD card it shows THERMAL RUNAWAY or POWER PANIC DETECTED randomly I also tried to go back to version 3.8.1-2869 but the same problem. Not sure if it is an electrical or software problem.

ChrisVahi, 2020-07-25:
I eventually found a broken bed thermistor wire while diagnosing this
problem. It was only failing during high y positions until I performed a
wiggle test which caused it to break completely. After repairing this
problem the to runaway errors ceased for me. I'm not sure if the new
firmware is more sensitive to bad thermistor wires but I haven't had any
trouble since I repaired it.

closed 2023-10-19 by github-actions[bot]