prusa3d/Prusa-Firmware #2509: [BUG] MINTEMP BED error causes menu button to not respond

clintonefisher, 2020-02-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**- 3.9.0-RC1

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

It's 10C in my workspace.  I get a MINTEMP BED error.  I press the selector knob button to open up the settings menu to preheat enough so I can reset and clear the MINTEMP BED error, but the button doesn't do anything.   Both a reset and power cycle leave me in the same state.

After having good MINTEMP recovery in 3.7.0 (rarely even having to reset), it looks like I'm back to getting the blowdryer out just to get back to printing.
  
**To Reproduce**
Put printer in 10C or lower temp environment.   Power it up.   Get MINTEMP BED error.  Press selector knob button.   Observe no action.  Reset, power-cycle, etc, ...no change.   Cannot do anything until bed is warmed by an external heat-source.
  
**Expected behavior**
Expected behavior is that there should be NO MINTEMP errors until heating is actually commanded, and even then, there should be an adequate check to insure whether it's just cold, or if a thermistor is actually faulty.

I thought that the long-time problematic and annoying MINTEMP behavior had been adequately addressed in 3.7.0, but it appears to be broken again.  
Many of us operate in shops that will be below MINTEMP until the enclosure warms up.   Having to get a blowdryer or heatgun out just to begin printing is not an adequate solution.

leptun, 2020-02-29:
Any severe error will prevent the user from using the printer until reset. The mintemp button blanking is expected behavior.
Can you please post some logs with the printer triggering mintemp? We dropped the mintemp temperature to 10C with 3.9.0-RC1. I'm curious how quickly it triggers the error.

OBELIKS, 2020-03-04:
The problem is that if you want to print with ambient temperatures that are at best a couple ° over outside temperatures, you are unable to print now. Now I have to heat the heatbed and nozzle with external source (my hands). The printer in in enclosure, so it is in warm place after a couple of minutes. Please think of a way to print even at low temps.
And it triggers immediately.

Naugrimohtar, 2020-03-13:
My basement is ~ 54C and every time now that I run my MK2.5s, when it starts out could it starts heating up and after a few minutes it gives min temp. I reboot and try again, no problems. It is a strong annoyance.

clintonefisher, 2020-03-14:
> Any severe error will prevent the user from using the printer until reset. > The mintemp button blanking is expected behavior.
I just need a way to start it up on a chilly morning, that's all.  I appreciate the lowering of MINTEMP to 10C, but preventing a pre-heat, even after a reset, means I cannot start printing without getting the blowdryer out, which is annoying.

> Can you please post some logs with the printer triggering mintemp? We > dropped the mintemp temperature to 10C with 3.9.0-RC1. I'm curious
> how quickly it triggers the error.
I don't typically keep logs (sorry), but I go to my printer in the morning and it says MINTEMP BED.   Then after a reset, it immediately shows MINTEMP BED and prevents any preheat or other menu actions.
Please take cold-starts into account in the MINTEMP logic:
After a reset, if temp is < 10C, allow preheat menu and on preheating just note it's a cold-start, check a timer and note temp changes.  If the temp doesn't get > 10C before the timer runs out, then MINTEMP error again and lock entirely (requiring a reset to retry).  
  I think that would keep everyone happy.  We just need to get the printer started once and then things stay warm enough to print happily. 
Thanks for all you do!

Naugrimohtar, 2020-03-17:
Here is a log file, hopefully what you are looking for
[MINTEMP Coldstarting MK2.5s.txt](https://github.com/prusa3d/Prusa-Firmware/files/4341294/MINTEMP.Coldstarting.MK2.5s.txt)

closed 2023-10-19 by github-actions[bot]