prusa3d/Prusa-Firmware #2510: [BUG] Bad handling of fsensor signal filament runout in Octoprint?

ochm, 2020-02-25:
I created issue in Octoprint, but maybe the problem is in printer's firmware, or compatibility between MK3S and Octoprint?
Whole behavior is described in https://github.com/foosel/OctoPrint/issues/3461

**Printer type** - MK3S

**Printer firmware version**-  3.8.1, 3.9.0-RC1

**To Reproduce** Print from Octoprint and let the filament sensor detect filament runout, change filament and try to continue print.

**Expected behavior**
The same behavior like M600 is in gcode or filament change is called from printer's menu.

leptun, 2020-02-29:
I was unable to reproduce the issue.
Things I tried:
- Filament change from menu
- Filament runout on MK3S
- M600 sent from octoprint terminal
All of them worked successfully.

ochm, 2020-02-29:
> filament runout in Octoprint? (#2510)

Strange, i can reproduce the issue it any time i want. 
Only problematic is resume print after filament runout. Others M600 states are handled well. 

​​-- 
Martin Och 
Mikrotik: MTCNA, MTCWE, MTCTCE 
Veeam: VMTSP, VMSP 
VMware: VCP6-DCV

ochm, 2020-03-05:
Problem solved! I turned on sending line numbering and checksums, now goes everything well.

closed 2020-03-05 by ochm