prusa3d/Prusa-Firmware #2511: [FEATURE REQUEST] MK3 Firmware to handle increased MMU filament count

cjbaar, 2020-02-26:
**Printer type** - MK3S
**MMU Upgrade** -MMU2S (modified)

**Is your feature request related to a problem? Please describe.**
I have been building a mod to the MMU to utilize more than 5 filaments (currently testing with 12). I am able to get a working prototype using the parts and MCU from the original MMU2 kit. I am also able to create a slice models successfully with Prusa Slicer, and I can see tools up to T11 (as expected) in the output gcode. However, the MK3 firmware seems to be hard-coded to ignore any tool changes greater than T4 (shown on-screen as F5). I have done debugging on the MMU side, and confirmed that when the gcode for a higher tool is reached by the printer, no tool change command is forwarded to the MMU, and the printer does not attempt an unload.
  
**Describe the solution you'd like**
Ideally, the firmware should send the gcode data as-is, without putting restrictions on the tool number. If necessary, the printer could query the MMU for the value of EXTRUDERS when it queries for firmware versions.

Panayiotis-git, 2020-02-26:
I'm really interested in your solution!

I recently did some prints with 16 different filaments and I did the changes by hand. An improvement was required to the firmware to keep filament changes under control:
https://github.com/prusa3d/Prusa-Firmware/pull/2380

I also have in mind to modify the MMU to something expandable (to multiples of 5?) utilizing the same controller.

Have you shared any of your designs?

vintagepc, 2020-02-26:
This is likely because internally we need to allocate memory for tracking filament depletion in things like spooljoin - currently fixed at 5 slots, of course. 
It would likely not be too hard to abstract those count checks into a #define in the variants. The main ones I'm aware of are in Autodeplete.h and the T-code handling. There are likely many more - LCD menus, Load codes, cut codes...

cjbaar, 2020-02-27:
I've started to look through the code, and see a lot of hard-coding around this. As you mention, in AutoDelete, there is a definition for number of filaments:
    `static const uint8_t filamentCount = 5;`
But, then, that variable does not seem to be used in other places that reference the MMU. For example, the T commands are parsed into an ENUM of all MMU commands, which I don't quite get, instead of using just "T" and a variable.

I am not super concerned with accessing all of the slots from the menu (ie, "load to nozzle"), just letting the gcode be understood and not truncated.

cjbaar, 2020-02-27:
@Panayiotis-git Thanks for the input and your PR; it's a nice improvement on the manual side. I will share more of the actual design once I can prove it works :)

vintagepc, 2020-02-27:
That's pretty cool; I contemplated chaining MMUs at some point for giggles if I happened to get another one for cheap from someone that no longer wanted it. 

Are you comfortable with making the changes yourself? If not I can probably spend a little time to help you out. No guarantees, but I think this is a worthy project to keep developing :-)

Panayiotis-git, 2020-02-27:
I wonder how you managed to fit the filament spools. They should take some place...

vintagepc, 2020-02-27:
> I wonder how you managed to fit the filament spools. They should take some place...

That's what samplers are good for... Chances are if you need a lot of colours in your print, you don't also need a ton of each. I could see  a setup where you have full spools of the main ones you use and just samplers or shorter lengths for the rarely used ones.

Panayiotis-git, 2020-02-27:
Good point! I hope slicer is precise.

vintagepc, 2020-02-27:
If not, that's what you have runout sensing for :P
 (Funny aside - I used to laugh at people with walls of filament when I first got my printer about a year ago. After some sales last November I'm starting to get just as guilty 🤣 )

cjbaar, 2020-02-28:
> I wonder how you managed to fit the filament spools. They should take some place...

Right now it's just all over my kitchen table. Haven't come up with good organization yet, or come up with a method to buffer them during MMU unload.

cjbaar, 2020-02-28:
> That's pretty cool; I contemplated chaining MMUs at some point for giggles if I happened to get another one for cheap from someone that no longer wanted it.

Yeah... "cheap" is the key word. I have been trying to minimize parts cost, and at this point, also avoid buying new motors, etc.

> Are you comfortable with making the changes yourself? If not I can probably spend a little time to help you out. No guarantees, but I think this is a worthy project to keep developing :-)

Thanks for the offer. I will probably dig a bit deeper at some point. I had to do some fair hacking on the MMU side which I didn't mind, but was hoping to avoid getting into the base firmware. I would probably try to submit it as a pull request, so need to make sure I don't break the existing functionality.

cjbaar, 2020-02-29:
@vintagepc I was able to get this working with (I think) minimal hacking. Essentially, I modified the following:
    1) Enable gcode characters outside of 0-4 (mod: 0-9)
    2) Extend manual ENUM commands for T,E,K,L from 0-4 to 0-31
    3) Change auto deplete storage from uint8_t to uint32_t
    4) Fix some LCD display issues with the double-digit filaments
I have tested with a range of filament->filament changes. I have not tested the auto-deplete, but the functions themselves remain untouched. If you have time, you can check out the changes here. Once I test a bit more, I will try to submit a PR.
    https://github.com/prusa3d/Prusa-Firmware/compare/MK3...cjbaar:MK3
[image]

vintagepc, 2020-03-01:
nice work - though at that scale it's probably time to rethink the CMD enum design... :-)

cjbaar, 2020-03-01:
> nice work - though at that scale it's probably time to rethink the CMD enum design... :-)
Agreed; but, for now, kept me from getting to deep with then changes.

> TBH I always felt like if it wasn't for the bondtech gears, the MMU could be a lot more compact and/or hold a lot more filaments... you've definitely proved that part 👍
Same here. I have experimented with several different designs over the last few months, and just recently came back to this which is based very heavily on the stock MMU. Finding compact gears was definitely the trick to getting more filaments out of the same space, and using the fixed 110m motor shaft.

cjbaar, 2020-03-01:
Hot off the presses... successful test print. 12 filaments, ~9 hours, 440 tool changes. (colors are not well-defined because I turned off purging)

[image]

cjbaar, 2020-03-01:
Created pull request https://github.com/prusa3d/Prusa-Firmware/pull/2515

cjbaar, 2020-03-01:
@vintagepc If you have a moment, I could use some help with the new variable I created (mmu_filament_count). It is declared in mmu.h, and assigned in mmu.ccp. It is used in AutoDeplete.cpp, and in Arduino IDE I had to re-declare(?) in AutoDelete.h as an extern. However, it is not passing the automated build checks on the PR. Is says "undefined references" in AutoDelete.cpp. Do you know where it is best to define this? Thanks.

vintagepc, 2020-03-01:
Since it is a quasi-constant and does not need to dynamically change during the program, I would make it a #define MMU_FILAMENT_COUNT in something like the variant .h file instead.

Design considerations aside, this looks like not an issue with the file itself but the test linkage/unit tests. Your changes altered the dependencies of the autodeplete function and the test needs to be updated as a result.

vintagepc, 2020-03-01:
Also, you need to undo your change to config.h before committing.

cjbaar, 2020-03-01:
Thanks. My goal was to keep it as a variable, in order that I can later come back and update it dynamically based on the startup queries to the MMU... thus if the MMU is changed or a different unit connected, the base firmware doesn't have to be re-compiled. Any thoughts on that?

vintagepc, 2020-03-01:
Well, TBH I think it's premature over-engineering. Adding a slot to an MMU (or even removing and installing it) is not a trivial activity. Longer term, sure, it makes sense to have something like that but I think the better architecture would be to actually ask the attached MMU how many filaments it supports on startup. (I heard rumblings about MMU comms improvements coming in 3.10 in another issue here) Having it live in the MMU itself makes a bit more sense than a property of the printer it is attached to, and achieves what you intended.

But that's just my 2 cents, I'm just a contributor here; you're completely welcome to ignore anything and everything I say. :-)

cjbaar, 2020-03-01:
Thanks again. Maybe I didn't frame it correctly, but what you said is exactly what I was trying to say. :) On startup, the printer queries the MMU for its version; I would like to eventually add the # of filaments to that query. That's why I wanted this to be a variable, and not a constant in the MK3 firmware. I am not familiar with how the tests are processed. I see the the test is including autoDeplete.h, but even when I had the variable declared in there it still failed. Do you think it would pass if I included mmu.h in the test.cpp?

vintagepc, 2020-03-01:
That's only part of it; your changes obsolete the current tests, so you really should also update the test itself to test your new functionality too.

cjbaar, 2020-03-01:
@vintagepc Thanks again for your assistance. I have updated the test suite for auto deplete, and was able to get it to run successfully on my local machine. For some reason, now my commits are not triggering the auto-build; maybe it had too many failures.

cjbaar, 2020-03-02:
FYI, another print from the new MMU, this one with 8 colors, plus a manual change to TPU at the higher layers.
[image]

cjbaar, 2020-03-04:
@vintagepc I did an empty commit and was able to force the test to rebuild, and it passed! Thanks again for all your input. Just curious, as I haven't contributed to a project like this before... are reviews done by Prusa staff, or just the community? As a contributor, is it good etiquette for me to look into other PRs I am interested in reviewing?

vintagepc, 2020-03-04:
Any approval and final merge has to be done by Prusa staff- it can't be a community free-for-all as they need to focus on support and features related to the stock hardware they sell. It also needs to be maintainable for them, since as soon as it is merged, they're basically taking responsibility for the feature going forward.

Something like this is likely pretty low on the radar in terms of supporting their primary hardware, so it may take a while. The 3.9 firmware is already defined and in the final stages of testing and merging the last few open marked PRs. I wouldn't be surprised if 3.10 was already mostly defined internally (but we have no visibility into their internal ticketing system apart from the occasional merge with a PFW-### reference somewhere...)

To your other question - it's not required, but if you have time you're welcome to check out or merge in PRs for other features, test them and report back. The more testing it gets, the better!

cjbaar, 2020-03-07:
@vintagepc The next phase of my project...
[image]

vintagepc, 2020-03-07:
Nice, is that using the MMU board? Does seem like a lot of untapped power on that.

cjbaar, 2020-03-07:
Yeah, at the moment. I did some digging through the schematics, and discovered that the unused port labeled "sensor" is just an exposed I2C bus, so the connections are easy. Biggest issue is that the graphic- and font-based libraries eat up a lot of program space, so I am using a more watered down version (https://github.com/greiman/SSD1306Ascii).

At some point, I might consider offloading this to another processor like and ESP, and then I could also try things like sending me a text when a load fails. But, then, I also to write another communication language.

JKinch03, 2020-06-29:
@cjbaar is there a place I can get the STLs and the firmware for the MMU mod?

cjbaar, 2020-07-02:
@JKinch03 I have posted the overview and files here: https://github.com/cjbaar/prusa-mmu-12x

closed 2023-10-19 by github-actions[bot]