prusa3d/Prusa-Firmware #2512: MMU2s direct motor control

los341, 2020-02-27:
Printer type: MK3S
MMU Upgrade MMU2S

Hello, I would like the ability to manually control the MMU2s extruder, idler, and selector motors, just like we can control the X, Y, Z, and E axis motors please.

Also, for the idler and selector motors, it would be nice to have the ability to adjust or fine-tune the positions for filament 1-5.

On more than one occasion, I've had to turn these axes manually.  It just seems like a logical feature enhancement to be added to the LCD control panel please.

leptun, 2020-02-28:
Because of the way the boards currently communicate, this is not really possible.
There were talks to have the MMU more “open” to the main board and act like a driver with virtual registers. This way the main board would have a lot more control over every aspect of the MMU unit, but that idea went nowhere for now.
MMU changes will be coming to 3.10.

los341, 2020-02-28:
That makes sense, understood.  I am however curious to know more about what's coming in 3.10.

closed 2023-10-19 by github-actions[bot]