prusa3d/Prusa-Firmware #2513: [FEATURE REQUEST] Time Remaining on Color

addisonefc, 2020-02-28:
I do a lot of prints with manual color changes and it would be awesome if the printer lcd rotated back and forth between total time remaining in the print and the time remaining for that color when it sends the M73. If it could do this it would be easier to plan how long until I need to be in person at my printer to execute the manual color change. PrusaSlicer is calculating these times already as evidence in some of that latest releases.

Example: My total print takes 8 hours. Color one takes 4 hours. Color two takes 2.5 hours. Color 3 takes 1.5 hours.

The LCD might rotate between something like this during color 1:

07:47R(emaining) 
03:47C(olor)

07:00R 
03:00C

04:01R 
00:01C

Color 2:

03:50R 
02:20C

01:35R 
00:05C

Color 3:

01:00R 
01:00C

00:07R 
00:07C

leptun, 2020-02-28:
You should also request the changes in PrusaSlicer (if you haven’t already) as currently the gcode does not contain the values in a readable format

Area5142, 2020-02-28:
I like the idea!
Doing some printing with colour change and pause I always has a note paper with times for next event.
I would be great if M73 M-Code was expanded to have information on time to the next pause event, like:

```
M73 P<percent> R<time> E<time to event> M <message to show>

M73 P1 R63 E24 M Colour Change
or
M73 P1 R63 E24 M Pause
```
The extension to M73 only tells the time to the next event and a short text to display. This would make it simple for the firmware and leave it up to PrusaSlicer to use for colour change, pause or any other event that needs manual intervention.

DRracer, 2020-03-02:
@leptun @Area5142 we are considering this feature both for PrusaSlicer and the 8bit FW probably for FW 3.10

The3Designer, 2020-06-29:
I would also like it if the display could indicate how much time is left untill there is a pause coded into the G-code file, for the same reason @addisonefc mentions. Since the 2.2 version of PrusaSlicer, we can now insert both filament changes and pauses at a certain layer/print height.

Perhaps, the LCD could even display if there is a pause or filament change during the print, indicated by a P or F respectively. I've made some quick mock-ups of how this could look like:

[image]
[image]
[image]

Note that I've moved the "Total print time remaining" to the place of the SD percentage, since I personally think that is not really useful information. This allows the "Time to pause" or "Time to filament change" to be positioned next to the "Total print time remaining" information.

If the SD percentage should remain present, then the "Total print time remaining" could indeed switch between the "Total print time remaining" and the first upcoming pause or filament change. Then, if there are no pauses or filament changes left, the LCD only needs to display the total print time remaining.

cimoalpacino, 2020-06-29:
I like your mod! Nice work.
If no pause or filament change, i would still like to see the percentage though. Otherwise i think info about pause/filament change has higher priority.

The3Designer, 2020-06-29:
@cimoalpacino 
Perhaps, if the SD percentage is still relevant/important to you and others, the total print time remaining (on the left in my mock-up) can switch between total print time remaining and the SD percentage, while the value on the right solely indicates how much time is left to a pause or filament change. If there are no pauses or filament changes left, having this time displaying "--:--" also indicates to the user that there are no pauses or filament changes left (i.e. that you can leave the printer alone from now on).

closed 2023-10-19 by github-actions[bot]